motioneye-project/motioneye #693: MotionEyeOs and V4L2 Camera on RaspBerry Pi3

Franz53, 2017-10-14:
Installation OK, camera detected. 
Motion detection works only once, than stops. 
If I change any parameter and do Apply, the first detection is saved (still images od videos) than nothing.
Other problem is that Automatic Brightness loops from dark to white continuosly. 
Some still images saved are B/W or Blue. 
 
If I add an USB camera, this second camera works fine. No problems. 

Here is motion.log

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-2.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2:ml2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

Thanks for any answer and sorry for my poo English. It's not my mother language. 

FF

Franz53, 2017-10-15:
Sorry! I see now thet the correct page for my issue is MotionEyesOs. 
About che camera, model is Pi NoIR Camera V2

JamBalaya56562, 2026-11-07:
Closing, as you noted above, this belongs to motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos). On current Raspberry Pi OS with motionEye, the camera module goes through libcamera, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack).

closed 2026-11-07 by JamBalaya56562