motioneye-project/motioneye #694: generating ssh keys

julzKB, 2017-10-17:
Hi there,

I'm trying to generate an ssh keypair but it's seems to be impossible to save it since "/root" is read only

`Saving key "/root/.ssh/id_rsa" failed: No such file or directory`

ssh-keygen is present as part of the OS but how would you guys create an ssh key pair ?

ccrisan, 2017-10-18:
What are you trying to achieve in the end? Do you want to have an ssh client running on your motionEyeOS machine to use those keys when authenticating with an external SSH server?

julzKB, 2017-10-19:
yes exactly , (or the other way around), generate public keys on the motionEyeOS and authenticate to an other ssh server in which case I run into this :

`Failed to add the host to the list of known hosts (/root/.ssh/known_hosts).`

The idea to is use rsync to send the files over to an other server .

ccrisan, 2017-10-19:
Generating keys for the local user is not currently supported.

The other way around, you could place your keys in `/data/etc`, as indicated by the `/etc/sshd_config`.

julzKB, 2017-10-19:
Fair enough, thanks for looking into it .

closed 2017-10-19 by julzKB

ccrisan, 2018-06-02:
I wrote [this article](https://github.com/ccrisan/motioneyeos/wiki/SSH-Client-On-motionEyeOS) to help users run SSH commands from motionEyeOS.