motioneye-project/motioneye #695: Motioneye is insecure by default (download last picture without user/pass)

Nekmo, 2017-10-19:
I have put a user and password in Motioneye, and when accessing the web interface, it is verified that the user and password are correct.

But if I download the last capture directly, no password is required.
```
[nekmo@homura /tmp]$ wget http://myserver:8765/picture/1/current/
--2017-10-19 04:20:58--  http://myserver:8765/picture/1/current/                                                                                                                                                                                  
Resolviendo myserver... 163.172.210.17                                                                                                                                                                                                            
Conectando con myserver[x.x.x.x]:8765... conectado.                                                                                                                                                                                        
Petición HTTP enviada, esperando respuesta... 200 OK                                                                                                                                                                                               
Longitud: 42126 (41K) [image/jpeg]                                                                                                                                                                                                                 
Grabando a: “index.html.3”                                                                                                                                                                                                                         
                                                                                                                                                                                                                                                   
index.html.3  100%[=========================>]  41,14K  --.-KB/s    en 0,09s    
```

Motioneye allows you to spy cameras without user and password!

I think my setup is correct.
```
@enabled on
@show_advanced off
@normal_password
@admin_username ********
@admin_password *******

webcontrol_html_output on
webcontrol_port 7999
setup_mode off
webcontrol_localhost on

thread thread-1.conf
```

There is an additional option in motioneye to further secure with http auth. But what good is it to make a login page, if you can skip the restrictions using wget? 

Some people might think it's just a "feature", but I SURE there are MUCH people with MotionEye instances on the Internet, thinking they are safe to have a login page. Why is it not CLEARLY said that security by default allows you to spy on the cameras?

I think this is a very serious problem that should not be taken lightly. We talk about cameras that allow people to spy on houses, and it is not clear that the default security is insufficient.

ccrisan, 2017-10-19:
Set a surveillance password and you won't be able to view the stream without authenticating.

closed 2017-10-19 by ccrisan

bulgin, 2022-03-14:
Could ccrisan clarify if establishing a remote http connection (via port forwarding) to motioneye on raspberry pi3+ login page for admin or user reveals the credentials in the wifi stream between internal router and raspberry pi3+ running a non-wired, wireless-only connection?  What about the http connection between remote user connecting to the login page?

I absolutely love this program and works really well, but a lot of posts about insecurity of http-only connections and although the surveillance password works from the point of login to the device, there are stream before that using http-only, also wireless.

Some clarity would be appreciated.  Thanks.

zagrim, 2022-03-14:
First of all, you should always open a new issue instead of piggybacking a closed one from 4 and half years back in time.

But anyway since this is very much related to the original topic of login security: If you have non-secure plain HTTP connection between the browser and server (here ME), it is by default clear-text and insecure regardless of the transport (wired or wifi) if there's any possibility to have someone listening on the transport (which in case of wifi could mean someone next door, or in the parking lot using a directional antenna, after they had gained access to your wifi network). 

Basically the only thing that can secure the login is to have the connection between the browser and the login page secured by TLS (HTTPS). I think there's been some discussion here on using a reverse proxy like Nginx or Apache in front of ME (preferably running on the same box as ME) to enable using a TLS certificate from Let's Encrypt, or something like that (with a quick search I found #438 which is rather old but on the point anyway).

MichaIng, 2022-03-14:
Probably we can add HTTPS support directly to motionEye, i.e. you can enable the HTTPS protocol and set a cert and key via `motioneye.conf`. This will require the need for additional dependencies, I bet, but when it is documented well in the config file, and HTTPS by default disabled, then we don't need to pull it automatically on module install.

Until then, yes, using a proxy with HTTPS enabled is the common way to do this for web applications. If the proxy is running on the same machine, you can bind motionEye to localhost (`127.0.0.1`) so that it cannot be reached from anywhere else than from the same machine (where the proxy is running): https://github.com/motioneye-project/motioneye/blob/2df0383/motioneye/extra/motioneye.conf.sample#L17-L19

Another thing we could do is applying a random (or hardcoded) password (for both, admin and surveillance) by default on first motionEye start. But users then need to check the process/service output accordingly (or it looses most of the purpose). This is quite common practice as well, both with a random password, printed on first start, as well as using a hardcoded password unless changed/set manually before first start.

However, indeed it makes sense to discuss such topics in dedicated new issues 😉.

bulgin, 2022-03-14:
> y-To: motioneye-project/motioneye <
> motioneye-project/motioneye ***@***.***>Cc: bulgin <
> [motioneye-project/motioneye] Motioneye is insecure by default
> (download last picture without user/pass) (#695)Date: Mon, 14 Mar 2022
> 06:49:03 -0700

the biggest issue for me is using a motioneye that is in a location but
connected to the home router via wifi.  That is where sniffing the wifi
stream will reveal credentials.  

https://github.com/motioneye-project/motioneye/blob/2df0383/motioneye/extra/motioneye.conf.sample#L17-L19

MichaIng, 2022-03-14:
As said, use a local proxy then.

bulgin, 2022-03-14:
Thank you Michalng.  This is very useful.  Sorry I'm new to this.
Do you have a link to instructions on how to setup the encrypted proxy?

07:03:50 -0700

bulgin, 2022-03-14:
This also looks promising using certificates:
https://robertopozzi.medium.com/have-fun-with-your-raspberry-pi-secure-your-application-with-self-signed-certificates-c8ef455a492f id="-x-evo-selection-start-marker">

MichaIng, 2022-03-14:
Use your favourite search machine, there are unlimited guide about this. At best use one that matches your distro version. Let's Encrypt / Certbot can be used to get a free public HTTPS certificate. If it's for LAN only, you can use a self-signed certificate as well, but browsers will always throw a warning where you need to confirm that you trust it.

This guide covers some more distros and looks good on first view for Nginx (I'd prefer over Apache): https://www.supereasy.com/how-to-configure-nginx-as-a-https-reverse-proxy-easily/
Ah your's covers the self-signed certificate generation. You can pick just this and apply those to the Nginx configs as well.

ccrisan, 2022-03-14:
@bulgin while I support the HTTPS idea 100%, in this particular case, your credentials will not leak as they don't leave your browser nor your motionEye server (they act like a shared secret). A signature is used instead to authenticate the request. Moreover, only a hash of your password is saved on both ME and your browsers local storage, so this is another small security "feature" to further protect your password.

:warning: However, this does not mean that your actual data (video stream!) is safe; on the contrary: it's going to be transmitted in clear "text" (unecrypted) between your motionEye server and your browser, effectively being sniffable by anyone intercepting this communication.

MichaIng, 2022-03-14:
Looks like I need to read and understand the code around this signature generation and verification. However, if the connection itself is not encrypted, whatever the client sends to the sever for authentication, whether a plain text password, a hash, token or signature, a man in the middle or sniffer can use it the same way. But at least it is then not trivially possible to authenticate via frontend/web UI directly, but only with crafted requests to the backend directly (as far as I can think of).

bulgin, 2022-03-14:
And this looks very, very promising - install instructions for
motioneye *with* nginx:
https://mpolinowski.github.io/devnotes/2019-02-07--motioneye-on-debian#installation-of-nginx-on-debian id="-x-evo-selection-start-marker">

13:48:15 -0700