motioneye-project/motioneye #696: Attached Pictures Time Span = 1 causes failures.

ioquatix, 2017-10-23:
Everything w.r.t. motion detection works except email notifications: short video, still images, email notification (with `Attached Pictures Time Span = 0`). When I set `Attached Pictures Time Span = 1` I get the following failure:

```
encoded 30 frames in 13.73s (2.18 fps), 70804.23 kb/s, Avg QP:5.68
x265 [info]: HEVC encoder version 2.5
x265 [info]: build info [Linux][GCC 7.1.1][64 bit] 8bit+10bit+12bit
x265 [info]: using cpu capabilities: MMX2 SSE2Fast LZCNT SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
x265 [info]: Main profile, Level-5 (Main tier)
x265 [info]: Thread pool created using 8 threads
x265 [info]: Slices                              : 1
x265 [info]: frame threads / pool features       : 1 / wpp(45 rows)
x265 [info]: Coding QT: max CU size, min CU size : 32 / 16
x265 [info]: Residual QT: max TU size, max depth : 32 / 1 inter / 1 intra
x265 [info]: ME / range / subpel / merge         : dia / 57 / 0 / 2
x265 [info]: Keyframe min / max / scenecut / bias: 10 / 250 / 0 / 5.00
x265 [info]: Lookahead / bframes / badapt        : 0 / 0 / 0
x265 [info]: b-pyramid / weightp / weightb       : 0 / 0 / 0
x265 [info]: References / ref-limit  cu / depth  : 1 / off / off
x265 [info]: Rate Control / qCompress            : CRF-5.0 / 0.60
x265 [info]: tools: rd=2 psy-rd=2.00 early-skip rskip tmvp fast-intra
x265 [info]: tools: strong-intra-smoothing lslices=8 deblock
x265 [warning]: specified frame type (3) at 0 is not compatible with keyframe interval
2017-10-23 13:53:44: [sendmail]    ERROR: Exception in callback <functools.partial object at 0x7f9e1dfcb788>
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 605, in _run_callback
    ret = callback()
  File "/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 369, in poll_process
    read_media_list()
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 347, in read_media_list
    media_list.append(parent_pipe.recv())
EOFError
x265 [info]: frame I:     21, Avg QP:2.96  kb/s: 168067.15
x265 [info]: frame P:     82, Avg QP:6.03  kb/s: 38034.68
x265 [info]: consecutive B-frames: 100.0% 
```

Any ideas what's going wrong?

ccrisan, 2017-10-23:
This looks like a duplicate of #685. @ioquatix are you on Arch Linux? Honestly I have no idea what's the cause of this error.

ioquatix, 2017-10-23:
Ah, yep this is a duplicate. It was working several months ago but it's not working now, and I also found the gallery is not working either. How can I help you debug this?

AZRadiohead, 2017-10-24:
So in the meantime, what's everyone doing to to view their saved media?  I've got Samba setup but it was so much nicer having it available in Motioneye. 

@ccrisan If there's anything else I can do to help get this fixed please let me know.  As I said in the other ticket, I'm running Motioneye on a Ubuntu 17.10 server.  I noticed you asked a few people if they were on Arch Linux, so obviously it's not limited to Arch.

ioquatix, 2017-10-24:
I'm using Samba too :p

ccrisan, 2017-10-24:
Guys, I'll have to find some time and investigate this. I'll do my best to get this done throughout this week, but I'm not promising anything.

ccrisan, 2017-10-29:
Fixed via https://github.com/ccrisan/motioneye/commit/64743308a7508e5744c9764a73489e0ed71afd53.

closed 2017-10-29 by ccrisan