motioneye-project/motioneye #697: Threshold percentage for motion detection

ioquatix, 2017-10-23:
I've been experimenting with some web cameras running at 2560x1440

The percentage pixels does not appear to work well for such high resolutions for some reason. I have a pretty wide (2.8mm) FOV.

When debugging motion capture, if I walk out my front door, about 20,000 pixels maximum change. It's only 0.005% of the total image pixels: 20000/(2560*1440), but the actual size of the region can be quite big geometrically.

I'm not sure how this could be set more reasonably, but it's tricky with the slider to set a low threshold value.

Perhaps it should be logarithmic, or a pixel threashold, or perhaps the motion detection isn't picking up enough pixels. I'll try to get some examples.

richri, 2018-02-18:
Hi,

I am facing a similar problem ... I want to set a threshold of only 2000px approx. This is a percentage that can hardly be set via the slider directly.

Maybe allowing to set an absolute pixel value as optional setting is a solution?

Best regards

MoonCactus, 2019-02-17:
Here is one "hackish" way to do it, but this is no permanent solution (it will be overwritten each time):
  * ssh to the device as `admin` (same password as in the UI)
  * make the root partition writeable: `mount -o remount,rw /dev/root /`
  * then edit `vi /usr/lib/python2.7/site-packages/motioneye/templates/main.html` and change the max and decimals in:
```
<tr class="settings-item advanced-setting" min="0" max="2" snap="0" ticksnum="5" decimals="2" unit="%">
<td class="settings-item-label"><span class="settings-item-label">Frame Change Threshold</span></td>
```
This is still not a pixels count, which would be impacted by any change to the camera resolution.
Though, might also set the `Text Overlay` to a custom string like `%D / %o` (i.e. number of live pixels changed / threshold set by the UI), as it will help a lot knowing your actual setting.

I also wish the slider in motioneye also hinted about the number of pixels that correspond to the percentage. Or better, that motion itself showed a percentage in the first place (which is not really motioneye business)!