motioneye-project/motioneye #698: Feature request: Add 'During Daylight'

rudiniemeijer, 2017-10-23:
If you're using MotionEye with a normal daylight camera, operations outside of daylight hours could mean recording black images during the night. It could therefore be an excellent addition for 'Working Schedule' to add an option to motion detect only during daylight. Because timezone is already present, calculating daylight should not be too big a problem.

Suggested working (all UI changes are in Working Schedule):
* The Detect Motion dropdown list would be placed as the first UI component under Working Schedule
* Working hours (monday to sunday) only appear (or is only accessible) when 'During Daylight' is not selected (or During Working Schedule or Not During Working Schedule _is_ selected)
* The user selects 'During Daylight' as an option in Detect Motion in the Working Schedule tab
* An informational text is shown for the current timezone (as was selected in General Settings)
* Preferably the current times for sunrise and sunset are shown
* Motion Detection only takes place on all days of the week from sunrise to sunset, as calculated for the region the timezone is selected for

dragonflyuk, 2017-11-02:
Even for those of us with ir lights, it would be useful to have 2 sets of brightness settings for day and night.

cmseal, 2017-11-19:
For the moment, try this:
https://blog.ligos.net/2016-04-18/Day-Night-Cycle-For-MotionEye.html

pkwentus, 2018-10-04:
+1 on this feature request.  I try to manage this issue by adjusting the working schedule but daylight savings time and the seasons really cause me trouble. 

I have a lot of problems with insects being attracted to the IR lights on my cameras.   I tend to get lots of nighttime pictures of these bugs making sweet love to the light.

dhuizinga, 2019-05-27:
Another +1, I am currently looking at time-lapses with a lot of pitch-black frames, this would be very welcome indeed

sunnenproducts, 2019-05-28:
My main goal is for surveillance while I'm away.   I live in the country and I get an excessive amount of pictures of bugs on my cameras. due to the IR light.   

This feature would be really nice because I end up having an excessive amount of photos at times.  2k+ at times.