motioneye-project/motioneye #699: latest motion screen: all pictures

tolitius, 2017-10-24:
unless I misunderstand something, whenever someone rings a bell at midnight I have no way to put my motioneye camera to use, since it takes about 40 seconds to wait for:

* login to motioneye
* click on the "Picture Browser" icon
* wait for at least 30 seconds (just a few days worth of pictures)
* scroll down, wait for thumbnails to populate to identify the one with a face

again, I could be missing something obvious, or perhaps a semi real time is not an intended use, but is it possible to add a separate, one click away, screen that would show `last N` pictures from the last motion?

chrisf4lc0n, 2017-10-24:
Are you using MotionEye or MotionEyeOS?

tolitius, 2017-10-24:
MotionEyeOS, should I post it there instead? Although I think this feature is "orthogonally useful" from both.

chrisf4lc0n, 2017-10-24:
I asked, because for MotionEye on Raspberry Pi I have got a workaround.

tolitius, 2017-10-24:
cool, yea I have both: MotionEye on one Pi and MotionEyeOS on another.

tolitius, 2017-10-24:
so what's a workaround?

chrisf4lc0n, 2017-10-24:
You have to build a Wireless Raspberry Pi Doorbell, once you have done that get back to me.

tolitius, 2017-10-25:
funny :)

@ccrisan I'd like to hear your thoughts. Is it not an intended usage for motioneye or there is something I am missing?

ccrisan, 2017-10-26:
@tolitius I honestly don't understand what your issue is about. I get that it takes some time to get to browse the media files but I don't see the connection with the door bell.

tolitius, 2017-10-26:
@ccrisan what I'd like to do is to minimize the time between someone ringing a bell and me knowing who that person is. If I use motioneye it takes me up to a minute to identify the person in front of my door.

Hence my suggestion is to have a separate screen that I can bookmark (i.e. say to a phone home screen) that would always show the `last N` images from the last motion detected.

This way when someone rings a bell, I can pick up my phone and see who it is within a couple of seconds.

ccrisan, 2017-10-26:
And what's wrong with the live streaming feature? It's as real-time as it gets.

chrisf4lc0n, 2017-10-26:
What about if you build a Wireless Raspberry Pi Doorbell and when someone presses the button it activities the script to send you an email with the last 5 frames? I have it working that way and it working flawlessly.

tolitius, 2017-10-26:
@chrisf4lc0n: (thought your were joking at first) I already have a pretty good doorbell (bluetooth) with its own mobile receiver. I don't think there is a need to build anything physical to solve this.

@ccrisan: ah.. most of the people a not staying on the spot where camera sees them or their face.
For example as a person walking towards the door the camera captures a lot of good images. Once the person approaches the door s/he stands next to the door bell where the camera can no longer provide any intel.

this is not the problem with a camera placement though, since given the layout it works great.. "but this one important use case"

ccrisan, 2017-10-26:
@tolitius while your proposal makes sense in some cases, we don't have any plans to implement such a functionality the near future. I'm afraid you'll have to choose one of the three variants (mine, @chrisf4lc0n's or your current one).

closed 2017-10-26 by ccrisan

chrisf4lc0n, 2017-10-26:
No mate I wasn't joking. The doorbell gets pressed Pi picks it up, runs an email script and attaches the 5 last movement frames sending it to myself. If you have got the doorbell connected to your Pi then you just need to get your head around the script, which I can help you with.

chrisf4lc0n, 2017-10-26:
I also ran another script, which you might be interested in. Doorbell gets pressed script opens up live stream in Chrome tab on my Ubuntu HTPC if it is on. It seemed to be delayed by about 5 seconds but it did the job.

tolitius, 2017-10-26:
@chrisf4lc0n sure, thank you for offering, your solution is good. But I really believe nothing physical needs to be built to solve this especially given the fact that I already have a great (non Pi) doorbell + raspberry pi which is my camera is also a turing complete piece of hardware / software that is fully "capable to render simple HTML with N images on port X".

I'll just do a file watcher and grab `last N` images myself and throw some CSS on it. It is not a hard thing to implement and I really believe there is a need: i.e. this is _not_ bloating the codebase for no good reason.

@ccrisan thanks, it would make sense to have this built in, I hope you'll have time sometime in the future.