motioneye-project/motioneye #700: GPU OR CPU

chrissierra, 2017-10-24:
Hello, I want to know the way to optimize motioneye with /boot/config.txt. 

Can somebody tell me how much  of gpu must I use? overclocking or not? 

etc. 

The most important question, Use motioneye GPU ? 16 or 64?

Thanks..

jasaw, 2017-10-24:
Presumably you're running motionEye on Raspberry Pi.

Whether you need to overclock the Pi or not depends on which model of Pi you're using. I recommend trying **without** overclock to see if performance is acceptable. Keep in mind that overclocking can potentially cause stability issue.

How much GPU memory is required depends on your configuration. Higher resolution needs more GPU memory. If you use OMX movie format, you're going to need even more GPU memory. Max video width and height of bcm2835-v4l2 driver also contribute to the GPU memory required.

Resolution: 800 x 600
Movie format: H264/OMX
bcm2835-v4l2 max_video_width=1920 max_video_height=1200
Minimum GPU memory required: 66MB

Resolution: 1280 x 720
Movie format: H264/OMX
bcm2835-v4l2 max_video_width=1920 max_video_height=1200
Minimum GPU memory required: 70MB

Resolution: 1280 x 1024
Movie format: H264/OMX
bcm2835-v4l2 max_video_width=1920 max_video_height=1200
Minimum GPU memory required: 75MB

Resolution: 1920 x 1080
Movie format: H264/OMX
bcm2835-v4l2 max_video_width=1920 max_video_height=1200
Minimum GPU memory required: 83MB

If you're not sure, 128MB is a good start.

chrissierra, 2017-10-24:
Ok, thanks. Please help me with some concepts, 
Can I choose that movie format "H264/OMX" or it is by default??
About BCM2835-v4l2: Where can I choose that options? ( max_video_width and max_video_height?)

jasaw, 2017-10-24:
H264/OMX is only available if your device supports it. Which OS and what version of ffmpeg are you running? You may have to recompile ffmpeg to enable OMX support. If you don't use OMX, GPU memory usage will drop. It's ok to allocate more GPU memory than what's needed.

BCM2835-v4l2 parameters can be set from one of the files in /etc/modules. Exactly which file depends on your OS, your set up.

chrissierra, 2017-10-24:
ffmpeg version 3.1.1 Copyright (c) 2000-2016 the FFmpeg developers
built with gcc 4.9.2 (Raspbian 4.9.2-10)
libavutil      55. 28.100 / 55. 28.100
libavcodec     57. 48.101 / 57. 48.101
libavformat    57. 41.100 / 57. 41.100
libavdevice    57.  0.101 / 57.  0.101
libavfilter     6. 47.100 /  6. 47.100
libswscale      4.  1.100 /  4.  1.100
libswresample   2.  1.100 /  2.  1.100
libpostproc    54.  0.100 / 54.  0.100

// I use raspberry pi 3, raspbian. 

BCM2835-v4l2 -> Only works if i use raspicam? Because I m using just usb webcams.

jasaw, 2017-10-25:
In that case, you don't need to worry about BCM2835-v4l2.

tigrrr, 2017-11-23:
To share my experiences with OMX, choosing H264 or Matroska OMX encoder, motioneye couldn't produce video files whereas regular H264/Matroska worked fine. GPU settings were 128 Mb, according to the recommendations above. Recorded video stream was RTSP 1280x720 plus another video of the same size being just monitored.

After increasing GPU size to 256 Mb, the problem is solved, there is a noteable drop in CPU load with OMX, which is great. I just wanted to share this solution, as it's not in line with the recommendations above, can someone explain this ?

JamBalaya56562, 2026-11-09:
jasaw answered this, and tigrrr found that H.264/OMX needed 256 MB of GPU memory there. Closing, as the question is answered.

closed 2026-11-09 by JamBalaya56562