motioneye-project/motioneye #703: events on motioneye

MichelSuire, 2017-10-26:
hello,
i hope you understand me, my (frenchEnglish is not very good) !
In motioneye, we see only two events: on_event_start and on_event_end. how can we add on_motion_detect and how to use on_area_detect ?
I think that on_area_detect is to see some move in a parts of the frame.
Perhaps it is not possible ? if it is, can i have one example ?
Merci beaucoup. (or more thanks! ... i think)

Michel Suire

chrisf4lc0n, 2017-10-29:
Sorry, I do not get it. What are you trying to achieve by adding on_area_detect and on_motion_detect?

MichelSuire, 2017-10-30:
I want to detect an event in a part of the frame and only if an other event arrive in the same time in an other part of the same frame, i want to start a signal. 
I d'nt know if it is possible with Motion?
Merci beaucoup
Michel Suire

SteveCohen, 2018-01-06:
Looking over the [official motion docs](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions), I think you need to use [area_detect](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionAreaDetect) to specify which area will trigger the script (specified by [on_area_detected](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionOnAreaDetected)) 

e.g. motion options like these would trigger `script.sh` if there was movement detected on the top of the frame (areas 1-3) OR on the left hand side of the frame (areas 4,7,9):
```
area_detect=123479
on_area_detected=/foo/bar/script.sh
```

ccrisan, 2018-01-07:
on_area_detect is not covered by motionEye but can be used, as @SteveCohen indicated, by adding those lines to the Extra Motion Options box.

closed 2018-01-07 by ccrisan