motioneye-project/motioneye #705: Downloaded video filename

themouseNZ, 2017-10-30:
When downloading video files for viewing, the downloaded file name should contain the camera name and date/time of the clip but this does not always happen for every camera.  I must be missing something but not sure what, I have checked and rechecked all settings but cant find anything related to this. Does anyone have an idea how to fix this behavior? Thanks

themouseNZ, 2017-10-30:
OK, I have found the reason, if there is a space in the camera name then the file name does not get the time added to the downloaded filename. Is this a bug or expected behavior?

ccrisan, 2017-10-31:
It surely sounds like a bug. I'll investigate it.

themouseNZ, 2017-10-31:
I replaced the spaces in the camera names with under-score's and now the file names download properly. Just as a quick question can you add the date to the file name as well as the time? Thanks

ccrisan, 2017-12-17:
I'm sorry but I wasn't able to reproduce this issue.

closed 2017-12-17 by ccrisan