motioneye-project/motioneye #707: recorded movies aren't sorted propery

arkenoa, 2017-11-01:
Please see attached image. [(https://s1.postimg.org/7y3uq0ddzj/Capture.jpg)](url)

jasaw, 2017-11-01:
You mean list is not sorted chronologically?

arkenoa, 2017-11-01:
Thank you for answering. That's right. The previous month newest events appeared on the top of the list, now I must scroll to the bottom to see November records.

ccrisan, 2017-11-01:
@arkenoa those are in fact folders with media files. Folders are sorted alphabetically (lexicographically) by motionEye. The default naming pattern (YYYY-MM-DD) will show the folders in chronological order when sorted alphabetically.

I can see that you have changed it to DD-MM-YYYY. When you sort folders using this naming pattern, you definitely won't get stuff displayed chronologically. My advice is that you stick with the default naming pattern.

closed 2017-11-01 by ccrisan