motioneye-project/motioneye #708: Not all videos uploaded

dragonflyuk, 2017-11-02:
We have our system on a 4g internet link. Which at times can be slow. However I have noticed that not all videos are uploaded to our Google drive, i think it's when the uploads yet back logged, some of the videos get lost.

dragonflyuk, 2017-11-02:
Btw the videos are still stored on the raspberry pi, just not uploaded to Google drive.

jasaw, 2017-11-05:
Could it be caused by the lack of upload retries? If an upload fails, motioneye doesn't try again.

jasaw, 2017-11-05:
I think upload retries has to be implemented properly in motionEye. @ccrisan Any thoughts on upload retries implementation?

ccrisan, 2017-11-06:
Guys there's no retry mechanism for uploading videos and implementing one wouldn't be the easiest task.

chrisf4lc0n, 2017-11-07:
If you are using MotionEye not MotionEyeOS you can try third party software for either Dropbox or Google Drive.