motioneye-project/motioneye #713: no static images showing up on centos7

donvipre, 2017-11-12:
Hi,

I was setting up a new centos 7 installation with motioneye behind a nginx and was wondering about the missing static images on the webpage.

I skipped nginx for now and tried to access motioneye without a proxy:

motioneye v0.38 is running on port 8765 under user motioneye and is listening to 0.0.0.0.
firewall has this port allowed.

For a http request i get the below in debug:
meyectl[22627]: DEBUG: 200 GET /static/img/motioneye-logo.svg (192.168.21.81) 0.56ms

downloading the file via direct link does work:
http://motion-host:8765/static/img/motioneye-logo.svg

I'm using Firefox on my client with version 56.0.

Is there any possiblity how to debug this or to even check whats going wrong?

I used motioneye for at least 2 years in a ubuntu container on the same machine.

I've uploaded my config files here:
https://paste.ee/p/PoCIq

all directories mentioned in motioneye.conf are belonging to motioneye user.

Disabling selinux did not help either.

Thanks in advance!

ccrisan, 2017-11-12:
Make sure you have the file `/etc/mime.types` installed on your system. On Arch Linux it's installed by `mime-types.`

donvipre, 2017-11-12:
Thank you very much!

that solved it!

for everyone with the same problem:
`yum install mailcap`

closed 2017-11-12 by donvipre

kubotan, 2020-06-18:
If it is cached, it will be loaded from the chrome disk cache, but at that time it seems that the content type may be application/octet-stream instead of Content-Type: image/svg+xml.
Removing the browser cache or specifying no-cache with set_extra_headers may help.