motioneye-project/motioneye #716: Unable to Start Motion on Ubuntu

am549, 2017-11-16:
Hi

I appear to be having the following error logged when trying to start MotionEye on ubuntu

meyectl startserver -c /etc/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.38
   ERROR: motion failed to start
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 332, in start_motion
    motionctl.start()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 132, in start
    raise Exception('motion failed to start')
Exception: motion failed to start
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 435, in main
    run()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 393, in run
    application.listen(settings.PORT, settings.LISTEN)
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1943, in listen
    server.listen(port, address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 142, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 197, in bind_sockets
    sock.bind(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

Any ideas?

ccrisan, 2017-11-16:
Yes, you probably started motion or/and motionEye twice. Once automatically and once manually.

am549, 2017-11-16:
Ah OK, I was trying to start it again as I'm unable to get my IP cameras to work. It doesn't load the feed any everytime I try and make a change I get the refresh error message. It looks like it's having issues authenticating with the IP camera for some reason but I can login fine from VLC with the same address, username and pw

JamBalaya56562, 2026-11-07:
Closing as answered: as ccrisan said above, "Address already in use" means motionEye was already running as a service, so starting it again by hand fails. The IP camera that would not load is a different problem; if it still happens with a current release, please open a new issue with `motion.log` and the URL format you use (without the password).

closed 2026-11-07 by JamBalaya56562