motioneye-project/motioneye #717: RTSP not reconnecting

martyouel, 2017-11-18:
Hi,

I'm using a new cam since 1 week and I have problem to keep it stable.

With my old cam, I was streaming from http.
With the new, I stream from RTSP feed.

One time a day, it looks like I loose the RTSP signal...which I think is normal on WIFI connection. The RTSP signal come back rapidly but motioneye do not reconnect. In fact, I think it's "motion behind" that do not reconnect. Did I post my thread at the good place ? Or I should post on motion project ?

I read an old post saying that we should use the latest version of motion but a version from "Mr-Dave". I tried this and it do not work. But i'm confuse, Mr-Dave seems to be the new owner of the motion project, so, should I rather get the latest latest version of motion ?

thanks in advance

kleini, 2017-11-18:
Which version of motion are you using? There was lately a 4.1 release of motion.

Just my two cents: Furthermore you should try to get the connection wired. Wireless connections are a mess for video streaming.

martyouel, 2017-11-19:
I did all the upgrade after posting my post. So i'm on motion 4.1.1. I also did an update of motioneye and i'm now on 0.38.

I know WIFI connection is not the best, but I bought a cheap camera that have only WIFI.

When the signal is lost MotionEye write a timestamp to know at what time the conneciton was lost. Anyone know where this information is stored ?
Maybe as a workaround I can build a script that do a monitoring of this stamp and do a restart of motioneye service when needed.

JamBalaya56562, 2026-11-07:
Closing, as this was with Motion 4.1.1 and motionEye 0.38 on a Wi-Fi camera, and there were no further reports. Connecting to the RTSP stream, and reconnecting after it drops, is done by Motion, not motionEye, and `motion.log` records when the connection is lost. If a current Motion still does not reconnect, please report it to [Motion](https://github.com/Motion-Project/motion) with that log.

closed 2026-11-07 by JamBalaya56562