motioneye-project/motioneye #720: Insert my motioneye homepage in my html homepage

bigpaolo, 2017-11-25:
Hy to all of you,
Hy ccrisan,
Motioneye is great!

My question:
I have insert the output of the cameras of motioneye in my html home page that has a secure access cover by a password.
I have used for each camera:

Http://user:password@192.168.1.10:8080/

And all is ok. 
I can wiev the output of the cameras (one for each port that Motioneye gives)in my home page.

But now i want to insert in my homepage the homepage of motioneye (the page that you have, when you enter with user and password (not with admin)) to see all the cameras in one output (because i have 7 cameras)

I have try:

Http://192.168.1.10:8081/

And I can import the Motioneye homepage but i have to insert user and password (THAT I DON’T WANT TO DO!) to enter.
I DON’T WANT TO LEAVE MOTIONEYE WITHOUT A PASSWORD!

I have try:

Http://user:password@192.168.1.10:8081/

But i receive a popup that says 
“The page don’t need a user and password. Pay attention...”

It seems that the Motioneye homepage don’t need security access, it’s only after that the access in Motioneye homepage that you have to insert your user and password.

Can you please help me to resolve this problem, to insert in my homepage the  Motioneye homepage to see all the cameras’ output ant to insert automatically my user and password.

Thanks 

Paolo

ccrisan, 2017-11-25:
If you want the camera frames embedded into your webpage without being asked for a password, you have two options:

1. Make sure you have no surveillance password set in motionEye. Then add the motionEye webpage as an iframe HTML element into your webpage. You'll be asked at most once to introduce your (surveillance) username.

2. You'll need to add each frame, manually into your webpage, partially creating the main view of motionEye. You'll be able to choose between the embedable frame view URL and the simple snapshot URL. The former acts as point (1) in terms of authentication, while the latter allows you to hardcode a signature in the URL so that each request automatically comes with required authentication.

bigpaolo, 2017-11-25:
Thanks ccrisan

As i described in my post, since today i do like you say in option n. 2.
But now i want to insert the Motioneye webpage but i cannot leave it without a password for obviusly security reason.
Are there any other way to hardcode a signature in the Url also for option 1 or:

Can you make please, in next Motioneye version the url to see all the own cameras instead to have theUrl for each camera?

Thanks in advance

Paolo

ccrisan, 2017-11-25:
That's not possible. You can't and you shouldn't be able to hardcode passwords/signatures for all possible requests in motionEye. I'm afraid you'll have to choose between the two variants, each one with its disadvantages, making the compromise that you find the most convenient.

closed 2017-11-25 by ccrisan

aizhesi, 2019-02-02:
What if you password protect a web page and in that page you embed the frames from your cameras?
password protect web page
leave user password blank on motioneye
?