motioneye-project/motioneye #721: OpenRC

kenny-wettlaufer, 2017-11-26:
Hey, I'm trying to get Motioneye to work on my Raspberry Pi 3B, running Gentoo ARM64, which uses OpenRC. How do I get it to work? I tried adding the sysv init file to the /etc/init.d but it gives me this error
I've got motioneye installed via pip2, when I try to add motioneye to the rc-update it says service not found.

/etc/init.d/motioneye.init start
/etc/init.d/motioneye.init: line 23: /lib/lsb/init-functions: No such file or directory

ccrisan, 2017-11-26:
The init scripts provided with motionEye are for older Debian versions and systemd-based OSes. I can't help you with any other init system. Please ask for help from the OpenRC guys as this has nothing to do with motionEye.

closed 2017-11-26 by ccrisan

kenny-wettlaufer, 2017-11-26:
:S where do I ask OpenRC guys??

ccrisan, 2017-11-26:
I didn't "drop" the sysvinit. I just added to the repo the init scripts that I knew about and had experience with. Anyone is welcome to contribute with other init scripts through PRs to the project.

kenny-wettlaufer, 2017-11-26:
I know, but I'm just saying its not just older versions of Debian that use SysV. When I get an OpenRC script to work with motioneye I will make a pull request, I'm making a post on the Gentoo forums right now.