motioneye-project/motioneye #723: OPCION NETWORK CAMARA NO APARECE EN MENU DESPLEGABLE

MarioMP1993, 2017-11-27:
Good morning, first of all, congratulations on the APP, I am going to raise a doubt that I have since I am not able to resolve.

when I install the system on a sd card, when adding a new camera, if I see the NETWORK CAMERA option in the pull-down menu, but when installing it on rapsbian I only see these two options, because it can be ?

On the other hand, when it comes to adding a hard drive to make the recordings the software does not detect it, there is a solution with respect to it?

Thank you.
[image]

ccrisan, 2017-11-27:
Make sure you have motion 4.1 installed.

closed 2017-11-27 by ccrisan

MarioMP1993, 2017-11-27:
Buenos días,

Gracias por la respuesta. Como puedo realizar la instalación de motion 4.1 en raspbian ? 

Me podrías pasar los comandos ?? 

Gracias! 

Mario

ccrisan, 2017-11-27:
You'll have to check out motion's [github page](https://github.com/Motion-Project/motion).

MarioMP1993, 2017-11-27:
Buenos días ,gracias por la respuesta.

Me podrías pasar por favor un link o como realizarlo?

Por otro lado, tengo otra duda porque el programa no me detecta un disco duro externo para poder realizar las grabaciones ??

Gracias otra vez por la respuesta. Necesitaría que me contestaste ya que sólo me queda ese paso y no consigo ver la opción NETWORK CAMERA en el menú desplegable.

Gracias

MarioMP1993, 2017-11-27:
Buenos días necesitaría por favor que me pasases cómo realizar la instalación de Motion, soy nuevo en esto y como le comento tengo todo instalado perfecto pero en el menú desplegable no me aparece la opción de network camera y las opciones de grabación tampoco me aparecen. Me sería de gran ayuda saber cómo instalar motion.

Saludos 

Mario

MarioMP1993, 2017-11-27:
Buenos días, de la versión 4.1 que paquete tengo que instalar en mi Raspberry y como? Que comandos tengo que utilizar, gracias por todo. 

[image]