motioneye-project/motioneye #725: Installing on Ubuntu cannot start service

pkwentus, 2017-11-28:
Rpi 3 install, Ubuntu MATE 16.04

In step #7 (part 2) of the Ubuntu instructions 
https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu

I keep getting the error: 
CP: cannot start '/usr/local/share/motioneye/extra/motioneye.systemd-unit-local' : no such file or directory

When I navigate to this location, I only see motioneye.conf.sample 

This is my 2nd time installing.  I removed partitions from the SD card, reinstalled MATE and followed the instructions above.

pkwentus, 2017-11-29:
I've tried re-installing Ubuntu 3x
I've tried a new memory card.  
It looks like the installation instructions or the content they are pullign have gotten corrupted.
Attached are screenshots of the termin window:

[image]
[image]
[image]

ccrisan, 2017-11-30:
Installing motionEye did not work because you don't have all the prerequisites: `curl` is missing. Make sure the motionEye installation succeeds before proceeding with the copy commands.

closed 2017-11-30 by ccrisan