motioneye-project/motioneye #726: Web control not working motion version 4.1

hinkma, 2017-11-28:
I update motion version 4.1 and then emulate_motion not working. Is this motioneye or motion problem?

landont, 2017-12-13:
I have tried to used simple commands in v4.0 such as:
curl http://userid:pw@127.0.0.1:7999/1/config/set?text_left=some_text

This no longer works in v4.1

ccrisan, 2018-01-05:
Hmm I'm not aware of any problems with the motion HTTP API in 4.1. Is this still a bug? If yes, can I see your configuration files (motion.conf, thread-*.conf) and what `curl` says when running that command?

hinkma, 2018-01-05:
Here is:
pi@RaspiServer:~ $ curl http://127.0.0.1:7999/1/config/set?emulate_motion=on
<!DOCTYPE html>
<html>
<body>
<h1>Not Valid Command</h1>
<p>The requested URL is not valid Command.</p>
</body>
</html>

motion.conf:

# @enabled on
# @show_advanced on
# @normal_password xxxx
# @admin_username admin
# @admin_password xxxx
# @normal_username user

webcontrol_html_output on
webcontrol_port 7999
setup_mode off
webcontrol_localhost on

thread thread-1.conf
thread thread-2.conf

thread-1.conf:

# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @motion_detection on
# @upload_port 
# @upload_location 
# @preserve_movies 7
# @network_username 
# @upload_movie on
# @id 1
# @manual_record off
# @upload_password 
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password 
# @upload_service ftp
# @name Autokatos
# @preserve_pictures 0
# @storage_device custom-path
# @manual_snapshots on
# @network_share_name 
# @upload_enabled off
# @webcam_server_resize off
# @working_schedule 
# @upload_username 

ffmpeg_output_movies on
height 720
netcam_userpass user:pass
stream_quality 85
threshold 17510
quality 85
noise_level 31
ffmpeg_output_debug_movies off
smart_mask_speed 0
pre_capture 10
noise_tune on
stream_maxrate 10
output_pictures off
stream_localhost on
ffmpeg_variable_bitrate 100
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
stream_port 8081
rotate 0
stream_auth_method 0
lightswitch 25
framerate 5
emulate_motion off
snapshot_filename 
despeckle_filter EedDl
snapshot_interval 0
minimum_motion_frames 10
stream_motion off
target_dir /var/lib/motioneye/Camera1
netcam_url rtsp://192.168.0.82:1024/11
text_double on
width 1280
post_capture 10
stream_authentication user:xxxxx
output_debug_pictures off
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left Autokatos
picture_filename 
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive off
max_movie_time 0
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
event_gap 30
netcam_tolerant_check on
mask_file /etc/motioneye/mask_1.pgm

All conf is like then is when I install and configurate motioneye.
If I change emulate_motion on and restart motioneye, start recording when motioneye start.

ccrisan, 2018-01-05:
Well in that case please [report a motion issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

closed 2018-01-05 by ccrisan