motioneye-project/motioneye #728: fast network camera not working anymore

dawauz, 2017-11-30:
Hello, I installed yesterday motioneyeos on a rpi 2b+ and csi cam.
I set it to fast network cam and it worked perfectly. I left it on this day, and now I dont have a camera anymore.
I can do settings, if I go back to normal cam, it is working again. What could this be? 
I already rebooted a few times and tried some settings but no luck. Red led is on, on the cam. It is not broken or something.

dawauz, 2017-11-30:
Ahw I was yelling to early, It wont work in IE11, but on chrome its giving me view.

JamBalaya56562, 2026-11-07:
Closing, as the live view works for you in Chrome and only IE11 failed. Fast Network Camera is a motionEyeOS feature, so please ask in the motionEyeOS repository about adding such a camera to another motionEye.

closed 2026-11-07 by JamBalaya56562