motioneye-project/motioneye #732: help me for my projects

Skydive1, 2017-12-04:
Hi everyone, 

I use motion version 4.0.1+git2a14Ofa ans OS version 20171008.
How to integrate script python or another langage on MotionEyeOs ?

Skydive1, 2017-12-04:
Here, my system works properly under my different Rpi (different images). However, I managed to write a code (inspired by several codes). I would like to actually integrate OpenCv with MotionEye, I do not know if it's feasible but I have not tested it yet. So, here are my different questions.
First of all, can opencv or simplecv be integrated with MotionEye?
Then, under MotionEye, how can we integrate a script on this system. Thank you

SteveCohen, 2018-01-06:
I imagine you could stream video from Motioneyeos to another computer running OpenCV.

ccrisan, 2018-01-07:
@Skydive1 this is a general question, not a valid motionEye issue. Please use the [Google groups](https://groups.google.com/forum/#!forum/motioneye) for this kind of questions.

closed 2018-01-07 by ccrisan