motioneye-project/motioneye #733: MJPEG over HTTP inconsistency

zocker1996, 2017-12-10:
When you are streaming the camera feed using MJPEG over HTTP there is some inconsistency:
The HTTP Resonse will give you
> Content-Type: multipart/x-mixed-replace; boundary=BoundaryString

as a header field, which is the correct mime type of MJPEG over HTTP.
However the frames are actually delimited by --BoundaryString and not as stated by BoundaryString.
Using a slightly different boundary string as claimed in the header seems to be a common misconception with MJPEG over HTTP. The boundary string should match the string in the header exactly.

ccrisan, 2017-12-10:
What motion version do you run?

zocker1996, 2017-12-10:
motionEye Version:    0.37.1
Motion Version:    4.0.1+git2a140fa
OS Version:    motionEyeOS 20171008

ccrisan, 2017-12-10:
The MJPEG stream is actually generated by motion. I recommend upgrading to motion 4.1 - it has many bug fixes and it may also fix this particular problem. If it doesn't, please [report a motion issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

closed 2017-12-10 by ccrisan