motioneye-project/motioneye #734: Install problem Raspbian on pi3

NikonD7200, 2017-12-12:
Hi all , i make a new installation on my pi3 and a take this message : 

_> pi@raspberrypi:~ $ sudo pip install motioneye
> Collecting motioneye
>   Downloading motioneye-0.38.tar.gz (756kB)
>     100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 757kB 277kB/s 
> Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye)
> Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye)
> Collecting pycurl (from motioneye)
>   Downloading pycurl-7.43.0.1.tar.gz (195kB)
>     100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 204kB 897kB/s 
>     Complete output from command python setup.py egg_info:
>     Traceback (most recent call last):
>       File "<string>", line 1, in <module>
>       File "/tmp/pip-build-sU6r9t/pycurl/setup.py", line 841, in <module>
>         ext = get_extension(sys.argv, split_extension_source=split_extension_source)
>       File "/tmp/pip-build-sU6r9t/pycurl/setup.py", line 508, in get_extension
>         ext_config = ExtensionConfiguration(argv)
>       File "/tmp/pip-build-sU6r9t/pycurl/setup.py", line 72, in __init__
>         self.configure()
>       File "/tmp/pip-build-sU6r9t/pycurl/setup.py", line 108, in configure_unix
>         raise ConfigurationError(msg)
>     __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
>     
>     ----------------------------------------
> Command "python setup.py egg_info" **failed with error code 1 in /tmp/pip-build-sU6r9t/pycurl/_**

Could you help me to find the problem ?

Thanks

ccrisan, 2017-12-12:
Have you run step 4 from [Install On Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)?

NikonD7200, 2017-12-12:
i find the problem , just make update and upgrade 
after make 
_apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev_
and finally
_pip install motioneye_

now Good !

electron1979, 2018-12-10:
Close issue???

closed 2018-12-17 by ccrisan