motioneye-project/motioneye #736: Problem : Not possible to do continuous video recording with stills on motion detect

grumpazoid, 2017-12-14:
motioneye 0.38
motion 4.1
raspbian 9.1

I am unable to continuously record video without having the stills also set itself to all frames.
I would just like stills when motion is detected. This way I can still send stills via email on motion but know I have full video recordings and motion triggered stills.
I set the stills to motion triggered and the video setting reverts to motion also.

Another request would be the ability to record at a low framerate and resolution continuously, but revert to a higher framerate and resolution on motion detection.

Thanks
G

granoviter, 2017-12-15:
Hi grumpazoid!

I probe use this continous recording but i have a problem :
https://github.com/ccrisan/motioneye/issues/735

Wich resolution use for a continuos recording? (I use full hd to hd)

grumpazoid, 2017-12-19:
Granoviter, I can record at full resolution (from IP camera). but I get frame drops.
Having researched this...it seems normal as the pi is not too capable at encoding.
I believe the dedicated pi camera should work better?

granoviter, 2017-12-20:
With RPi cam work well. (with right settings) Full HD + ~12 fps

homesmarty, 2018-01-15:
I ran into the same issue - when you switch video recording to continuous, the same happens for photos - so you either waste a lot of space having video and photos for each frame, or can't send mails with photos attached - which is really sad.
The fact that there are two different settings (for photos and videos), but they change each other silently, is very confusing.

ccrisan, 2018-01-16:
@homesmarty unfortunately that's the way with motion: you either use motion detection or you don't. Continuous recording is achieved by telling motion to permanently consider that motion is detected. You can't have that only for videos unfortunately.

homesmarty, 2018-01-16:
Thanks for your reply, Crisan!
I think I found a workaround, although not very elegant: Define to cameras! I'm actually using two different streams of my cam and this seems to work. I'll probably switch to record video in VGA continuously, while taking motion triggered snapshots in a higher resolution. But on a RasPi 3b framerates are already really low (~6 fps) for VGA + 720p streams...

jasonhpchu, 2018-08-31:
Can someone clarify this for me, I'm currently on motion detection only, thinking of switching over to continuous recording of video.
If I turn on continuous recording, I won't be able to use the motion detect function?
I technically don't need snapshots during motion detect, but if there's a way I can write a text file with the filename properly timestamped, then I can quickly eyeball when something happened and look at the video file.

ccrisan, 2018-08-31:
As i already said, continuous recording is achieved by simulating permanent motion. So no, you can't have both.

jasonhpchu, 2018-08-31:
OK, thanks.
Guess only way is to do what homesmarty said, use two streams, but that seems like a bit overkill on the traffic.

ITCMD, 2020-12-02:
> Continuous recording is achieved by simulating permanent motion.
Does this mean that turning off the motion detection swtich will disable continuous recording?

starbasessd, 2020-12-02:
No, it ignores motion detection and just sets the trigger to be 'always on'.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?