motioneye-project/motioneye #737: Debian - Motioneye not running when SMB_share true

tnikiel82, 2017-12-15:
Hello, 

I installed the motioneye on Debian 9. I did it according to the instructions on your project.

When I set SMB_share in motioneye.config and restart program then not running properly.

Where is the problem  - how I should run motioneye as root??

Regards, 
tnikiel

ccrisan, 2017-12-15:
Yes, you must run it as root when enabling smb share management.

closed 2017-12-15 by ccrisan

tnikiel82, 2017-12-16:
Hi,
yes,  I know this but I don't know how :)
Maybe some hint?

Regards, tnikiel.

ccrisan, 2017-12-16:
@tnikiel82 you simply start your motionEye server as root. If you're on systemd, it already runs as root. If not, it's up to your startup method to execute the command as root.

tnikiel82, 2017-12-16:
Thanks, a lot!

Problem was diferent - In log I found - install **cifs utils** :)

When I added this pack everything work:)

Regards, tnikiel