motioneye-project/motioneye #738: Add new text overlay

MatejKovacic, 2017-12-17:
I have bought temperature and humidity sensor SHT-3x and connected it to my RaspberryPi Zero W. I am using simple python script for reading temperature and humidity.

Now I would like to show temperature and humidity as an overlay text on video. Right now you can add just two overlays (left and right text), which could be only predefined (camera name, timestamp or static text).

It would be nice to have an option to add not just a static text, but also some text which could be read for instance from a file at a given time period.

In that case I would be writing temperature and humidity to a file, and MotionEye would read that data from a file each 10 seconds (for example) and put an overlay to a video.

What do you think of the idea?

ccrisan, 2017-12-18:
That could be achieved using motion's HTTP API and modifying the text_overlay option on the fly. However there's support for that in motionEye - you'll have to make requests to motion yourself.

MatejKovacic, 2017-12-18:
Hmm, could you point me to documentation for that? I was looking around and have found just a post describing how API will be implemented, but it is stated it is not yet...

ccrisan, 2017-12-18:
I was referring to [motion's API](https://motion-project.github.io/motion_config.html#Options_Stream_Webcontrol). You probably found [motionEye's API Draft](https://github.com/ccrisan/motioneye/wiki/API-(Draft)) which is indeed simply a draft for now.

ReliableBob, 2019-12-25:
I am also interested in this question.
For example, display over the video stream the values of MQTT sensors.
Please show some examples of use

jasaw, 2020-01-10:
Example (I have not tested this):
Overlay Raspberry Pi temperature in video. Have a cron job call this script periodically to update the overlay text.

```sh
#!/bin/sh

CAM_CONFIG_FILE="/etc/motioneye/camera-1.conf"
CAM_NAME=''
if [ -f $CAM_CONFIG_FILE ]; then
	CAM_NAME=`grep text_left $CAM_CONFIG_FILE | cut -f2 -d' '`
fi
TEMP=`vcgencmd measure_temp | cut -f2 -d'='`
curl "http://localhost:7999/1/config/set?text_left=$CAM_NAME\%20$TEMP"
```

ReliableBob, 2020-01-10:
> Example (I have not tested this):
> Overlay Raspberry Pi temperature in video. Have a cron job call this script periodically to update the overlay text.
> 
> ```shell
> #!/bin/sh
> 
> CAM_CONFIG_FILE="/etc/motioneye/camera-1.conf"
> CAM_NAME=''
> if [ -f $CAM_CONFIG_FILE ]; then
> 	CAM_NAME=`grep text_left $CAM_CONFIG_FILE | cut -f2 -d' '`
> fi
> TEMP=`vcgencmd measure_temp | cut -f2 -d'='`
> curl "http://localhost:7999/1/config/set?text_left=$CAM_NAME\%20$TEMP"
> ```

This code really works!
Thanks a lot!

davidcryptic, 2021-02-24:
I was having trouble using a new-line with the HTTP API, in case it helps anybody else, use \\n in Python:

import requests
requests.get("http://localhost:7999/1/config/set?text_left=new\\nline")

Is there a way to change what font is being used? Or perhaps place a black box behind all of the text? Some numbers are difficult to decipher at a glance if the background is a light color. See example (is the DegF 9.84? 9.04? 9.34? It's not immediately obvious):

[image]

thallaa, 2021-03-20:
Thanks for these tips. I went on a bit further and made a Python script which listens for temperature updates over mqtt and updates the overlays. This is obviously made for my own use but I thought I should throw it out here if someone wants to do something similar.

The script is ugly since I'm not very fluent with Python but perhaps somebody wants to make it less ugly.

But here you go: https://github.com/thallaa/motion-temperature

davidcryptic, 2021-03-22:
I decided to go a different route and just used "raspistill" to capture jpg images using the timelapse argument ("--timeout 0 --timelapse 5000" to capture an image every 5 seconds until you kill the process). I was also able to capture at the full resolution of the camera with raspistill (jpg captures are faster than png captures).

Then I used Python and PIL to make my own text overlay:

`from PIL import Image, ImageDraw, ImageFont`

`dateTimeText = filename.split('_')[0] + '\n' + filename.split('_')[1].split('.')[0].replace('-',':') #2021-03-22_01-06-15.jpg`
`tempF_RH_Text = "DegF: " + str(round(shelvingAverageDegF, 2)) + "\nRH: " + str(round(shelvingAverageRH, 2))`

`img = Image.open(filename)`
`d = ImageDraw.Draw(img)`
`font = ImageFont.truetype('DejaVuSansMono.ttf', size=40)`
`d.text((3030, 2360), dateTimeText, fill='white', font=font, stroke_width=2, stroke_fill='black')`
`d.text((10, 2360), tempF_RH_Text, fill='white', font=font, stroke_width=2, stroke_fill='black')`
`img.save(os.path.join(outputFolder,filename))`

Here's the result, which I find much easier to read:
[image]

thallaa, 2021-03-23:
Nice, that font looks much more clear.

MatejKovacic, 2021-03-23:
Would be possible that this (much clearer) font would be default? Or that we would be able to select font type in config?

starbasessd, 2021-03-23:
Tagged as 'enhancement request'

zagrim, 2021-03-24:
Regarding the font, that is not possible to change via Motion config, so the only way to influence that part is to make a feature request to Motion project.

Everything in the comments up to Jan 2020 deal with a sort of a "hack" to dynamically alter the camera config and that way updating the overlay data, which happens pretty much outside of MotionEye (which only supports static text and whatever dynamic content [Motion](https://motion-project.github.io/motion_config.html#text_left) enables to be used via conversion specifiers). I'm not quite sure but it would seem logical that also the font size decrease mentioned in [this comment](https://github.com/ccrisan/motioneye/issues/738#issuecomment-784750708) was due to Motion, not MotionEye. Looking at the ME code, the only way `text_scale` config param gets from ME side is by the GUI "Text Scale" slider.

zagrim, 2021-03-24:
@MatejKovacic the only way I see how [davidcryptic's raspistill based code](https://github.com/ccrisan/motioneye/issues/738#issuecomment-804254722) could be adapted to MotionEye is to set something like that to post-process the saved pics by adding the script as an "end command" in case you're using motion detection, or standalone in the background monitoring for new files to be produced. I may be missing something here, though.

tomholland, 2021-05-18:
On the idea of being able to specify a clearer 'font' for the text overlay; looking at how this is implemented in Motion itself, it's not a font being specified - the characters are each being drawn using a fixed size grid of pixels defined for each character:

https://github.com/Motion-Project/motion/blob/release-4.3.2/src/draw.c

You can see the grid values for each character defined there; a 0 is no pixel, a 1 is a black pixel, and a 2 is a white pixel.

To have it use a specified font instead would not be a small change, I suspect; the best suggestion I would have (for someone with better C skills than myself) would be using a larger grid for the character definitions (and that would then require adjusting the output for scale). Whilst not being perfect, it would allow for clearer characters.