motioneye-project/motioneye #742: Purge files based on disk usage

ShanDestromp, 2017-12-20:
In my use case I would rather ALWAYS record and delete based on file usage rather than time.  Sometimes a motion event just doesn't trigger; or something might happen in a masked area that I normally wouldn't care about but for some reason now do; and I'd like to have the ability to see it.

Unfortunately with my setup; the file sizes just aren't consistent enough to go just by age, and maximize my available storage.  Periods of high activity tend to make files much larger than when there is very little changing in the scene; nighttime footage is typically smaller than daytime, etc.  This means I have to either err on the side of caution and set a timer short enough to never "go over", or risk losing recordings because I've run out of space.

As such my "gift" to the project. 

It's a very quick and dirty bash script.  I don't know JS or Python otherwise I would have tried to integrate it in directly.  As it stands now it could feasibly be attached to an event (eg anytime a new file would be stored) or simply run via cron periodically.

It does require that all cameras record into one parent directory.  Sub dirs are fine eg:
> /opt/camera/channel1, 
> /opt/camera/channel2/YYYY-MM-DD

however this won't work:

> /mnt/camera1
> /opt/camera2

Or at the very least, it will only work on one of the directories.  It would be relatively simple to pass  variables to the script over-riding things however if you wanted it to operate on a per-channel basis.

The script will only do anything once you're **already** using more than your declared amount, and will only bring you down TO that amount (or rather, the first file deleted that brings you under it).  This means it wont stop you from running out of room if you set the limit to your max available storage; you'll want to leave it low enough for buffer space before the next trigger happens.  As it sits, it deletes one file (plus associated .thumb) at a time, checks the space again, and repeats until it gets below that limit.  At the end, if you've emptied a directory; it'll delete that as well.

It would be simple as well to modify and have a "start" and "stop" limit on it.  EG trigger when you've hit 5TB used, delete until there's only 4TB used; you'd just have to add an extra variable and change line 30 to compare $SIZE against that lower limit.

I parse the motioneye.conf file for two variables ("max_size" in MB and "media_path" for the parent directory of the recordings); but I've also set it where you can ignore those and just declare those settings inside the script.

I tried to keep it as generic as possible for cross-platform use but I can't make any promises that it'll work on anything other than my own system :-)

[disk_clean.sh.txt](https://github.com/ccrisan/motioneye/files/1574002/disk_clean.sh.txt)

tehKaiN, 2018-01-01:
I've got myself such functionality with quicker & dirtier approach. It just checks if disk usage is greater than 95%, if so removes recordings with hysteresis to 90% of disk usage:

```sh
#!/bin/sh

cd /mnt/6c5c6e37-de46-4411-a55e-46467f4111ed/motioneye
if [ `df -B1 /mnt/6c5c6e37-de46-4411-a55e-46467f4111ed | tr -s ' ' | tr -d % | tail -n 1 | cut -d ' ' -f 5` -ge 95 ] ; then
        while [ `df -B1 /mnt/6c5c6e37-de46-4411-a55e-46467f4111ed | tr -s ' ' | tr -d % | tail -n 1 | cut -d ' ' -f 5` -ge 90 ] ; do
                find -name "*.*" -type f -printf '%T@ %p\n' | sort -k 1nr | sed 's/^[^ ]*//' | tail -n 1 | xargs echo
                find -name "*.*" -type f -printf '%T@ %p\n' | sort -k 1nr | sed 's/^[^ ]*//' | tail -n 1 | xargs rm -f
        done
fi
```

jezmck, 2018-07-15:
This would be a great feature. I don't know how long it's going to take to fill my SD card, and it's unlikely to match 1 day/week/month anyway.

MattWeiler, 2023-10-24:
This would be a great feature.
It's the only thing keeping me on BlueIris.

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #302, which asks for the same: delete the oldest media when the disk gets full, rather than by age. The disk usage script above is a workaround until then.

closed 2026-11-13 by JamBalaya56562