motioneye-project/motioneye #743: motion not detected

FlyingPersian, 2017-12-20:
Hi
As mentioned in this post, https://github.com/ccrisan/motioneyeos/issues/897 , I have problems making motioneye detect any motion. I've installed Raspbian on a RPI 3, installed motioneye according to the wiki and most of the functions seem to work fine. I can't get it to detect motion and automatically take pictures/videos. 

Here my configs:

motioneye.conf: https://hastebin.com/vaxewomota.vbs
motion.conf: https://hastebin.com/ahaponiloj.nginx
motion.conf in motioneye dir: https://hastebin.com/edokohobuq.rb

Not sure, which motion.conf is being used, so I pasted both. Should be the first one though, since I turned on logging in the first motion.conf and I have the motion.log file in /var/log/

ccrisan, 2017-12-20:
How about thread-1.conf, which is in fact the one that matters? Make sure to remove any sensitive data from it.

FlyingPersian, 2017-12-20:
Sorry, I wasn't aware of the importance of the thread-1.conf. there should not be any sensitive data as it's mostly all in default settings for now

https://hastebin.com/acusaravox.pl

FlyingPersian, 2017-12-21:
I think I got it to work. I  changed the motion settings to 10s, 1 frame, 1 frame, 2 frames, and then it worked. Are there any recommended settings? I changed it to 10s, 30frames, 30 frames, 2 frames, and the video was 68s long, although the actual movement was only about 2s. I'd like it to catch couple of seconds before and after the motion. thought 30 frames are 1 second, since I set the frame rate to 30 fps.

geoffp13, 2019-07-02:
I have created a thread elswhere about motion detect not working in the hope of finding an answer to this very frustrating problem. Mention is made above about "thread-1.conf" ?? cannot find that !! another mentioned "zdump" - that does not exist either.
To see all the detail of my issue its in the other thread but not sure how to link it here. It seems as long as you use the software WITHOUT changing any settings it works as provided however it gives the wrong time locally (UTC) and motion detect floods the SD card with images and video so its pretty much useless really. Tried on a Pi 4 today and its just the same !!
I am very happy with the Linux operating system having used it for years so I know the problem is not there. Something in the software is coded wrong but I cannot find the source code.
I have a Windows 10 laptop with YawCam installed and it works perfect with the Pi Zero W and camera so its not a hardware problem however I don't want to leave it on all day and night.
Anyone help please ??

zagrim, 2019-07-03:
@geoffp13, a generic friendly advice: If you have a problem with any software, please open one issue for that in the issue tracker of that project and don't piggy-back others' (old) issues which may not be related at all.

Regarding the source code, you're already on github, so just click on the Code tab :smiley: above (unless you need sources for motion which has its own github project you surely can also find).

JamBalaya56562, 2026-11-07:
Closing as solved, as you got motion detection working by changing the motion settings to 10 s, 1 frame, 1 frame and 2 frames. Please open a new issue if it stops working again.

closed 2026-11-07 by JamBalaya56562