motioneye-project/motioneye #744: Action button 500 error

sharkoz, 2017-12-20:
Hi,

When I define action buttons, they appear on the cam overlay.
When I execute the scripts directly from ssh, they work OK. They are executable, I chmod 777 them.

But when I click on the buttons, I get a 500 error :  {"error": "internal server error"}

I don't know why it doesn't work, I can't find any log that would explain why.

I am on version 4.0.1

Thanls

ccrisan, 2017-12-21:
Check the motioneye log file for details. This has nothing to do with motion.

alc6379, 2018-03-25:
I'm having the exact same issue. The action I'm trying to do is just a shell script, but I get this: 
OSError: [Errno 8] Exec format error

Again, same thing as OP, the shell scripts themselves work fine if I run them from bash/sh, but I get the 500 error and the message above.

tictacbluw, 2018-04-09:
I got the same problem.
My buttons appears on motioneye hub but when I click on them I got "An error occured. Refreshing is recommanded"
Motion eye log : 2018-04-09 14:36:08: [motioneye]    ERROR: 500 POST /action/1/up/?_=15232845678$
Any ideas ?
PS : with ssh if I execute my script up_1 with "./up_1" they all work fine

closed 2018-04-09 by ccrisan

Egorkat, 2018-04-10:
Hi, I have almost similar problem - should I open new Issue or can post here?

ccrisan, 2018-04-10:
@Egorkat post your motioneye log here.

ccrisan, 2018-04-10:
Please send your motioneye log file (often obtained using `journalctl -u motioneye > motioneye.log`)

Egorkat, 2018-04-10:
ups, sorry - No journal files were found, if I run that command. I'm using Debian Jessie

Egorkat, 2018-04-10:
made this log file, sorry for misunderstanding
[motioneye.log](https://github.com/ccrisan/motioneye/files/1894245/motioneye.log)

ccrisan, 2018-04-10:
Wait, what? Are you trying to stop the motionEye server from a request made to it? I don't think you can do that and expect a response to the request.

Egorkat, 2018-04-10:
Got idea from scripts in https://github.com/ccrisan/motioneye/issues/203
Now I see that it may be impossible to run under motioneye...

shooftie, 2020-10-04:
Just out of interest @ccrisan, is it possible to add a script for enabling and disabling motion detection via action button?

Ideally it would be great if there was a global setting for enabling on all cameras, however, an action on each would be great.

ccrisan, 2020-10-05:
> Just out of interest @ccrisan, is it possible to add a script for enabling and disabling motion detection via action button?

I don't see why not. As long as you have your script that does your enabling/disabling motion detection, just give it the corresponding action button's name.

shooftie, 2020-10-05:
Thanks for your speedy response. Sorry, I should have been more descriptive.

The only way that I can see to programatically enable/disable motion detection is to edit the config file `/etc/motioneye/camera-1.conf` by changing `@motion_detection {on|off}` and then reloading MotionEye `systemctl restart motioneye`.

Is this correct or is there a better way?