motioneye-project/motioneye #745: No MMAL camera detected

roger-zacharczyk, 2017-12-21:
When adding new camera and choosing Local MMAL Camera there is no camera in the Camera drop down list. It says "(no cameras)". 

I'm using Arch on Raspberry Pi Zero W. Raspistill works fine. It works when I choose Local V4L2 Camera with Camera set to mmal service 16.1. Although, 2 FPS is way too low... This is why I want to add it as MMAL Camera.

ccrisan, 2017-12-21:
[This](https://github.com/ccrisan/motioneye/blob/master/motioneye/mmalctl.py#L24) is the function that's being called when listing MMAL devices. What it basically does is finding the `vcgencmd` full path and executing it with the argument `get_camera`. If the result says `detected=1` and `supported=1`, the function will report the presence of exactly one *VideoCore Camera*. The camera will then appear in your UI if it's not already present.

The first thing to do is to run the following command and see what it has to say:

    vcgencmd get_camera

roger-zacharczyk, 2017-12-21:
[root@rpi ~]# /opt/vc/bin/vcgencmd get_camera
supported=1 detected=1

However...

[root@rpi ~]# vcgencmd get_camera
-bash: vcgencmd: command not found

Then I thought you expect this to be accessible without specifying the full path.

So...

ln -sf /opt/vc/bin/vcgencmd /usr/bin/vcgencmd

And now the camera gets detected.

ccrisan, 2017-12-21:
Hmm so Raspbian doesn't have the vcgencmd in its default $PATH. I'm wondering why. I haven't personally used Raspbian in a long time, since I rely solely on [thingOS](https://github.com/ccrisan/thingos/)-based systems. Nevertheless, I'll update the wiki page accordingly.

roger-zacharczyk, 2017-12-21:
> so Raspbian doesn't have the vcgencmd
It's Arch I'm using. ;)

roger-zacharczyk, 2017-12-21:
BTW is there a way to enable FNC if I only use motioneye instead of motioneyeos? I don't see this option there.

ccrisan, 2017-12-21:
I really need to start reading the entire issue before commenting :)

ccrisan, 2017-12-21:
> BTW is there a way to enable FNC if I only use motioneye instead of motioneyeos? I don't see this option there.

There might be a way but you're better off using mjpg-streamer-based tutorials available online. Basically you'll need to get [streamEye](https://github.com/ccrisan/streameye) running on your Pi. You won't get the UI part - that's motionEyeOS-specific and it starts becoming too difficult to pull it out and use it standalone with a motionEye server.

closed 2017-12-21 by ccrisan

jmlsywl, 2023-12-26:
why do i have problem such as (no camera) when selecting Local MMAL Camera. i got confuse since when i instruct command "libcamera-hello -t0" in terminal, my camera function correctly. but when i command vcgencmd get_camera it resulting support=0 detected=0 libcamera interface=0?

MichaIng, 2023-12-27:
Much has changed in the 6 years since this issue was closed. Please check recent open issues about MMAL/DSI cameras or in case camera module 3 and alike on modern RPi systems.

jmlsywl, 2023-12-27:
Mr., let me introduce my self, my name is Jamal, currently going thru for
my degree in university in Malaysia. For me, I am actually just new with
this raspberry pi. Haven't learn about anywhere. I can't even understand
when I read and refer on website or utube. But my lecturer ask me for my
final project work, and to ensure I can graduate on time, I need to create
on cctv system by using raspberry pi with motioneye detection. I have been
struggle for 3 month with no result at all. Too much of things that I don't
understand  and no lecturer that know how to use this raspberry pi. So, I
just hope that, Is there any chance for you to teach me from scratch and
help me out to finish this project? Please?

MichaIng, 2023-12-27:
As said, you will be more successful checking some recent issues. See e.g. here how others got it working using `libcamerify`: https://github.com/motioneye-project/motioneye/issues/2812