motioneye-project/motioneye #747: Jessie to Stretch Upgrade, pip install motioneye fails due to lack of zlib headers

jonatversprite, 2017-12-22:
Documenting in case others encounter this error.  Following the instructions https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian and received an error when installing motioneye. I had to install some zlib packages to make it work. I added several packages, so I'm not sure which exact one fixed it.

`pi@raspberrypi:~ $ pip install motioneye
Collecting motioneye
  Downloading motioneye-0.38.tar.gz (756kB)
    100% |████████████████████████████████| 757kB 230kB/s 
Collecting jinja2 (from motioneye)
  Downloading Jinja2-2.10-py2.py3-none-any.whl (126kB)
    100% |████████████████████████████████| 133kB 728kB/s 
Collecting pillow (from motioneye)
  Downloading Pillow-4.3.0.tar.gz (13.9MB)
    100% |████████████████████████████████| 13.9MB 9.9kB/s 
Collecting pycurl (from motioneye)
  Downloading pycurl-7.43.0.1.tar.gz (195kB)
    100% |████████████████████████████████| 204kB 872kB/s 
Collecting tornado>=3.1 (from motioneye)
  Downloading tornado-4.5.2.tar.gz (483kB)
    100% |████████████████████████████████| 491kB 257kB/s 
Collecting MarkupSafe>=0.23 (from jinja2->motioneye)
  Downloading MarkupSafe-1.0.tar.gz
Collecting olefile (from pillow->motioneye)
  Downloading olefile-0.44.zip (74kB)
    100% |████████████████████████████████| 81kB 1.8MB/s 
Collecting backports_abc>=0.4 (from tornado>=3.1->motioneye)
  Downloading backports_abc-0.5-py2.py3-none-any.whl
Collecting certifi (from tornado>=3.1->motioneye)
  Downloading certifi-2017.11.5-py2.py3-none-any.whl (330kB)
    100% |████████████████████████████████| 337kB 358kB/s 
Collecting singledispatch (from tornado>=3.1->motioneye)
  Downloading singledispatch-3.4.0.3-py2.py3-none-any.whl
Collecting six (from singledispatch->tornado>=3.1->motioneye)
  Downloading six-1.11.0-py2.py3-none-any.whl
Building wheels for collected packages: motioneye, pillow, pycurl, tornado, MarkupSafe, olefile
  Running setup.py bdist_wheel for motioneye ... done
  Stored in directory: /home/pi/.cache/pip/wheels/2e/68/14/d669357ec459318c3c84b3d5a5950bc8f6458ae380bac228e5
  Running setup.py bdist_wheel for pillow ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-raRs7i/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpLBE28epip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/PIL
  copying PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/_util.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/ImageCms.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/CurImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/version.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/ImageMath.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/ImageMorph.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/OleFileIO.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/TarIO.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/ImageDraw2.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/ImageDraw.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/GimpGradientFile.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/ImagePalette.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/MspImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/PyAccess.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/PaletteFile.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/_binary.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/FontFile.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/TiffTags.py -> build/lib.linux-armv7l-2.7/PIL
  copying PIL/SunImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
[... 270 lines cut ...]
pi@raspberrypi:~ $ pip install motioneye
Collecting motioneye
Collecting jinja2 (from motioneye)
  Using cached Jinja2-2.10-py2.py3-none-any.whl
Collecting tornado>=3.1 (from motioneye)
Collecting pillow (from motioneye)
  Using cached Pillow-4.3.0.tar.gz
Collecting pycurl (from motioneye)
Collecting MarkupSafe>=0.23 (from jinja2->motioneye)
Collecting singledispatch (from tornado>=3.1->motioneye)
  Using cached singledispatch-3.4.0.3-py2.py3-none-any.whl
Collecting certifi (from tornado>=3.1->motioneye)
  Using cached certifi-2017.11.5-py2.py3-none-any.whl
Collecting backports-abc>=0.4 (from tornado>=3.1->motioneye)
  Using cached backports_abc-0.5-py2.py3-none-any.whl
Collecting olefile (from pillow->motioneye)
Collecting six (from singledispatch->tornado>=3.1->motioneye)
  Using cached six-1.11.0-py2.py3-none-any.whl
Building wheels for collected packages: pillow
  Running setup.py bdist_wheel for pillow ... done
  Stored in directory: /home/pi/.cache/pip/wheels/49/97/22/638a2ad41d85bc151da7c987ae644da429a23e95733fabfd76
Successfully built pillow
Installing collected packages: MarkupSafe, jinja2, six, singledispatch, certifi, backports-abc, tornado, olefile, pillow, pycurl, motioneye
Successfully installed MarkupSafe-1.0 backports-abc-0.5 certifi-2017.11.5 jinja2-2.10 motioneye-0.38 olefile-0.44 pillow-4.3.0 pycurl-7.43.0.1 singledispatch-3.4.0.3 six-1.11.0 tornado-4.5.2
`

NikonD7200, 2018-01-03:
i think you may have to use sudo : 

"pip install motioneye"

JamBalaya56562, 2026-11-07:
Thanks for documenting this. Closing, as it was about the Python 2 install on Raspbian Stretch; for anyone installing now, the current steps are in the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md).

closed 2026-11-07 by JamBalaya56562