motioneye-project/motioneye #749: motion_4.1-1 installation problems

parmiani, 2017-12-23:
Hello, 
i tried to install the latest version of motion but i encountered an error:

pi@raspberrypi:~ $ sudo dpkg -i pi_stretch_motion_4.1-1_armhf.deb
Selecting previously unselected package motion.
(Reading database ... 115711 files and directories currently installed.)
Preparing to unpack pi_stretch_motion_4.1-1_armhf.deb ...
Unpacking motion (4.1-1) ...
dpkg: dependency problems prevent configuration of motion:
 motion depends on libavcodec57 (>= 7:3.2.8) | libavcodec-extra57 (>= 7:3.2.8); however:
  Package libavcodec57 is not installed.
  Package libavcodec-extra57 is not installed.
 motion depends on libavdevice57 (>= 7:3.2.8); however:
  Package libavdevice57 is not installed.
 motion depends on libavformat57 (>= 7:3.2.8); however:
  Package libavformat57 is not installed.
 motion depends on libavutil55 (>= 7:3.2.8); however:
  Package libavutil55 is not installed.
 motion depends on libswscale4 (>= 7:3.2.8); however:
  Package libswscale4 is not installed.

dpkg: error processing package motion (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (215-17+deb8u6) ...
Processing triggers for man-db (2.7.0.2-5) ...
Errors were encountered while processing:
 motion

Someone has  this problem?

Thanks
Alessandro

ccrisan, 2017-12-23:
This is not a motionEye issue. Please open a [motion](https://github.com/Motion-Project/motion) issue.

closed 2017-12-23 by ccrisan

kleini, 2017-12-23:
Just installing the missing packages should solve the issue. Something like
`apt install libavcodec-extra57 libavdevice57 libavformat57 libavutil55 libswscale4`
should do the trick.

parmiani, 2017-12-24:
Thanks for the suggestion.
I tried but seems the package is not available:

pi@raspberrypi:~ $ sudo apt install libavcodec-extra57 libavdevice57 libavformat57 libavutil55 libswscale4
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libavcodec-extra57
E: Unable to locate package libavdevice57
E: Unable to locate package libavformat57
E: Unable to locate package libavutil55
E: Unable to locate package libswscale4

Alessandro