motioneye-project/motioneye #750: Motioneye causes rpi3 to hang following motion detection

duckulaj, 2017-12-28:
[motion.zip](https://github.com/ccrisan/motioneye/files/1590130/motion.zip)
RaspBerry Pi 3

Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"

VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

sudo systemctl status motion
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion; generated; vendor preset: enabled)
   Active: active (exited) since Wed 2017-12-27 23:49:16 UTC; 38min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 343 ExecStart=/etc/init.d/motion start (code=exited, status=0/SUCCES
   CGroup: /system.slice/motion.service

Dec 27 23:49:13 raspberrypi systemd[1]: Starting LSB: Start Motion detection...
Dec 27 23:49:16 raspberrypi motion[343]: Not starting motion daemon, disabled v
Dec 27 23:49:16 raspberrypi systemd[1]: Started LSB: Start Motion detection.

sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor prese
   Active: inactive (dead) since Thu 2017-12-28 00:20:41 UTC; 8min ago
  Process: 354 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/m
 Main PID: 354 (code=exited, status=0/SUCCESS)

Dec 27 23:49:40 raspberrypi motion[638]: [1:ml1] [WRN] [ALL] mlp_retry: Retryin
Dec 27 23:49:40 raspberrypi motion[638]: [1:ml1] [ALR] [VID] v4l2_start: Failed
Dec 27 23:49:40 raspberrypi motion[638]: [1:ml1] [ERR] [VID] vid_start: V4L2 de
Dec 28 00:20:38 raspberrypi systemd[1]: Stopping motionEye Server...
Dec 28 00:20:38 raspberrypi meyectl[354]:     INFO: interrupt signal received, 
Dec 28 00:20:38 raspberrypi meyectl[354]:     INFO: server stopped
Dec 28 00:20:38 raspberrypi meyectl[354]:     INFO: tasks stopped
Dec 28 00:20:41 raspberrypi meyectl[354]:     INFO: motion stopped
Dec 28 00:20:41 raspberrypi meyectl[354]:     INFO: bye!
Dec 28 00:20:41 raspberrypi systemd[1]: Stopped motionEye Server.

The above systemctl status are taken after issuing sud systemctl stop motioneye.

I can boot the raspberry pi 3 and it will run until motion is detected. The CPU usage peaks at around 85%. The system then becomes unresponsive and a hard reboot is required. This only happens when motion is detected. I have run motion without motioneye and there is no problem.

Any assistance would be aprreciated.

duckulaj, 2017-12-28:
I have now attached motion.log just after raspberrypi halted again.

Hope that helps

[motion.log](https://github.com/ccrisan/motioneye/files/1590683/motion.log)

ccrisan, 2017-12-28:
Does the motion completely crash? Can you watch your memory usage around that event? The out-of-memory killer may take it down (it'll be visible in your dmesg).

duckulaj, 2018-01-14:
Apologies for the delay in responding as I have been on holiday. I have now moved motioneye to a Linux Mint machine and having no problems.

closed 2018-01-14 by duckulaj