motioneye-project/motioneye #752: event_image_snapshot: Could not create symbolic link

sbourdette, 2017-12-30:
1. version : motion 4.1.1 motioneye 0.38
2. installed as a package : deb 
3. standalone or part of third party : motioneye 
4. video stream source : net cam rtsp 
5. hardware [x86, ARM, other]: virtualbox x86
6. operating system [Linux (which), FreeBSD, other]: ubuntu 16.04

I have a continous message in my log for my 3 net cams : 
 `[1:ml1] [ERR] [EVT] event_image_snapshot: Could not create symbolic link [/timelapse/20171230/20171230_214235.jpg]: Read-only file system`
`[2:ml2] [ERR] [EVT] event_image_snapshot: Could not create symbolic link [/timelapse/20171230/20171230_214236.jpg]: Read-only file system`
`[3:ml3] [ERR] [EVT] event_image_snapshot: Could not create symbolic link [/timelapse/20171230/20171230_214236.jpg]: Read-only file system`

the FS is not in read only

conf in motioneye.conf
`# default output path for media files (must be writable by motionEye)`
`media_path /media/sf_Survey/motion`

directory 
`sylvain@motion:/media/sf_Survey$ ll`
`total 40`
`drwxrwx--- 1 root vboxsf  4096 déc.  28 19:43 ./`
`drwxr-xr-x 4 root root    4096 déc.  27 17:18 ../`
`drwxrwx--- 1 root vboxsf  4096 nov.  19 14:53 events/`
`drwxrwx--- 1 root vboxsf 16384 nov.  25 15:08 lost+found/`
`drwxrwx--- 1 root vboxsf 12288 déc.  30 20:33 motion/`

group :
`sylvain@motion:/media/sf_Survey$ grep motion /etc/group` 
`video:x:44:motion`
`motion:x:120:`
`vboxsf:x:999:sylvain,motion`

SteveCohen, 2018-01-06:
Can you confirm the directories you're trying to write to? From the output it looks like you're trying to writing snapshots to the ``/timelapse...`` but your other files are all in ``/media/...``?

sbourdette, 2018-01-07:
The problem wasn't a motioneye issu but a Virtualbox issue.
The pictures are stored on a vboxsf and motion was unable to create symlink.

Symlink creation is disable on vboxsf by default.

closed 2018-01-07 by sbourdette