motioneye-project/motioneye #753: Problem connection

NikonD7200, 2018-01-01:
Hi all , today my connection to motioneye don't work but nothing change in my configuration
MotionEye answer : Invalid credentials. 

Raspberry pi3 + motion + motioneye 

How to fing the problem ?

NikonD7200, 2018-01-04:
I find i password deseppear , it's link new installation 

very strange

tokamac, 2019-05-03:
Same problem here. Hassbian Strech, installed motion 4.2.2 fine, then motioneye with pip2 (to avoid issues with Python3). Worked like a charm until I installed wrong `/dev/video*` for my USB (V4L2 H.264) camera. Was immediately unlogged, and from this moment on I can never log in MotionEye GUI. When I type my ID and password I get "invalid credentials". If I enter again my password, a new blank window opens with the address `http://MYI_P/login/`
I uninstalled and reinstalled motioneye, rebooted each time, with or without the camera connected… motioneye is totally broken. 
Any solution? For example can we erase the file containing that password? Where is it located?

tokamac, 2019-05-03:
This is the same issue as in MotionEyeOS : https://github.com/ccrisan/motioneyeos/issues/1839

JamBalaya56562, 2026-11-08:
Closing, as this was the Python 2 release (pip2, Raspbian Stretch), and the second report points to motioneye-project/motioneyeos#1839 for the same problem. Login works differently since 0.44.0 (#3332), with sessions instead of signed URLs. If a correct password is still refused with a current release, please open a new issue with the motionEye log from the failed login.

closed 2026-11-08 by JamBalaya56562