motioneye-project/motioneye #754: Can not get BananaPi to boot from image

mage00000, 2018-01-04:
I have tried several versions of motioneyeos:
I use the supplied writeimage script (only renamed it) on linux.

[root@localhost Banana PI]# ./writeMEyesOSimage -i motioneyeos-bananapi-20180101.img -d /dev/sdd -n SinBelle:xxxxxxxx -s 192.168.1.210/24:192.168.1.1:8.8.8.8
:: writing disk image to sdcard
220+0 records in
220+0 records out
230686720 bytes (231 MB, 220 MiB) copied, 20.8475 s, 11.1 MB/s
:: re-reading sdcard partition table
:: mounting sdcard
:: creating wireless configuration
:: setting static IP configuration
:: unmounting sdcard
:: you can now remove the sdcard

But the images do not boot. Only the red (power) led lights up....
also tried: motioneyeos-bananapi-20170329.img and motioneyeos-bananapi-20171008.img
same problem.
I use a 16Gb SD card which works fine if I put a fedora image on it, but the motioneye images just don't work (for me).

Any idea how to proceed?

ccrisan, 2018-01-04:
What Banana PI board variant do you have? Can we see a picture of it?

mage00000, 2018-01-04:
Thanks for your quick response!
I've got a M3 V1.2 , see image:
[image]

ccrisan, 2018-01-04:
Unfortunately only Banana PI M1 is currently supported by motionEyeOS.

closed 2018-01-04 by ccrisan

mage00000, 2018-01-04:
OK, thanks for the info!
I'll get another board and try again!