motioneye-project/motioneye #755: E: Package 'ffmpeg' has no installation candidate

chrissierra, 2018-01-04:
E: Package 'ffmpeg' has no installation candidate

I cant install motioneye; i have done before but installation page was edited and now it dont works.

ccrisan, 2018-01-05:
What distro?

chrissierra, 2018-01-05:
Raspbian ; on raspberry pi .

ccrisan, 2018-01-05:
Then make sure you're running the latest Raspbian (based on Debian 9 Stretch).

chrissierra, 2018-01-05:
**Ok, so one oldy distro that run apt-get update & upgrade isnt the same that the latest ?**

Motioneye works in the same way in:

RASPBIAN STRETCH WITH DESKTOP and  RASPBIAN STRETCH LITE ??

ccrisan, 2018-01-05:
As long as `lsb_release -a` says "Debian Stretch" (or "Debian 9" or so), you should be fine.

kleini, 2018-01-05:
@chrissierra `apt upgrade` does not upgrade a Debian Jessie to Stretch. That requires a different upgrade command.

chrissierra, 2018-01-05:
If i use motion ; and set framerate and streaming framerate at max values with an 3 US$ orange pi zero of 256mb ram etc, so simple; it display an amazing framerate.... Question: If i use motioneye; motion config (motion.config)  will  influence on motioneye framerate ???

ccrisan, 2018-01-05:
@chrissierra I don't really understand your question, but if you're asking if motionEye has an impact on the framerate, it does, but not necessarily. If you're watching the MJPEG stream generated by motion (you could do that with motionEye), then you get the exact same FPS.

Anyways, I think this issue can be closed now.

closed 2018-01-05 by ccrisan