motioneye-project/motioneye #756: WARNING: Connect error on fd 16: ECONNREFUS causing many false alarms

homesmarty, 2018-01-06:
I am struggling to get motion detection to work and suspect it's a technical problem... I see a lot of warnings in the motioneye.log:
WARNING: Connect error on fd 16: ECONNREFUS
They come in groups (~5-30) and always coincide with false alarms of motion detecion (video recorded, but nothing moving).
What does this warning mean? How can I dive deeper to fix the issue?

I get dozens of recordings of nothing, but when I walk though the picture, nothing is recorded :(
I'd love to use motionEye as a NVR (Wansview W2 ZIP camera, might get more of those), it's much slicker than Synology Disk station ;-) But right now, I can't get it to do the job at all...

Versions I use:

motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008
On a RasPi3

PS: This might or might not be a motionEye issue, but I need help - is there a forum where I could seek help?

Zortris, 2018-03-31:
Same issue here

[motion.log](https://github.com/ccrisan/motioneye/files/1865658/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/1865659/motioneye.log)

Zortris, 2018-04-27:
Push

JamBalaya56562, 2026-11-07:
Closing, as this was with motionEye 0.37.1 and the Motion of that time, and there were no further details. That warning means motionEye could not connect to Motion's stream of the camera, which happens while Motion restarts or reconnects to the camera, and the false alarms at the same time likely come from that too. If both still happen with current releases, please open a new issue with `motion.log` from around one of the false alarms.

closed 2026-11-07 by JamBalaya56562