motioneye-project/motioneye #758: Optimal Hardware for Multiple High Quality Streams?

DevScientist, 2018-01-10:
Hello all,

I originally started out my project using RPis and motioneyeOS but decided I wanted higher quality streams. So I set up a virtual machine on my Windows Desktop running CentOS7 and installed the motioneye service. (3/4 CPU cores on my i5-3570K and 4GB RAM are currently allocated to the VM)

I have three network cameras streaming to the motioneye server via RTSP/UDP. These cameras are capable of streaming up to 1080P at 20+FPS, so no hardware bottleneck there. With this setup and no motion detection (although I want it) enabled I am getting 720P at about 10 FPS on all three cameras. Which is not bad at all...But I want to make it better. 

I plan on building a dedicated Linux home server that will host motioneye. My question is: What kind of hardware specs would I need in order for motioneye to handle 6 RTSP streams at 720P 25 FPS and processing motion detection on 3 of them? I know this is relatively taxing on the CPU, but is there a way to make an estimation so I don't spend too much?

Thank you so much in advance. This has been my first big endeavor into Linux systems and networking at it has been so much fun. This is some excellent software you've created!

DevScientist, 2018-01-10:
Found a previous question in which ccrisan answered and said he had no idea what hardware specs would be necessary. Guess I'll just buy the fastest I can afford and see what happens!

closed 2018-01-10 by DevScientist

kleini, 2018-01-10:
It mostly depends on motion what can be processed. motionEye has nothing to do with that except the web frontend.
At home I have a Core i7 3820 which is not enough for two 4k (2688x1520) streams at 25fps. The latest updated Docker images with motioneye are based on my image creation improvements. I try to integrate there motion with GPU support, which should then be enough power for high quality streams.

DevScientist, 2018-01-10:
Hi kleini,

So this might be a question better posed to the motion team?

Ah, so if I can leverage some processing power from the GPU it could help smooth things out. How would I go about doing that? Sorry, still relatively new to this but learning fast. 

Thanks

kleini, 2018-01-10:
GPU support is not really good documented. motion uses FFmpeg libraries and those can be built with GPU support: https://trac.ffmpeg.org/wiki/HWAccelIntro
I have no idea, how motion uses FFmpeg libraries and whether is is possible there to gain more speed using the GPU.

jasaw, 2018-01-10:
Hi guys,

To enable GPU accelerated video encoding, first you need to find out which codec is the GPU accelerated version. To get motion to select the codec, you'll need to modify motion config option. For example, to select h264_omx codec, you specify: `ffmpeg_video_codec mkv:h264_omx`. I pushed this feature to upstream motion but the documentation wasn't updated. One more thing, h264_omx codec has been blacklisted by motion software by default because of a bug in unpatched ffmpeg.

GPU accelerated video decoding is possible too, but I haven't got time to submit a push request to upstream motion. I have a proof-of-concept that works, just need time to make it into a push request.

Hope that gives you guys clues on how to enable GPU acceleration.

TopperBG, 2018-01-11:
Follow, it's good topic.