motioneye-project/motioneye #761: Unable to open video device

chrissierra, 2018-01-12:
Hello, I want to know if it is fixable that situation: 30 min says Unable to open video device; another 30 min works normally, an so on...  Its a motioneye problem? a motion problem? A Device's issue? 

Please, help. .

ccrisan, 2018-01-12:
It's definitely not a motionEye problem and it's probably not a motion problem either. It's most likely a device/driver or hardware related issue. What you could do is:
 * make sure the power supply is enough
 * make sure your system is up-to-date
 * reduce the capturing resolution and/or frame rate (which in turn will reduce used bandwidth)
 * try with a different camera (model)
 * try other OSes

These are general ideas, given that you did not disclose anything about your setup.

closed 2018-01-12 by ccrisan

chrissierra, 2018-01-24:
@ccrisan  There is a way to know ( with crontab or another way ) programatically that a usb webcam gives that error : Unable to open video device.

While a month there is no problem, but suddenly gives this error.   I must to restart motioneye and reboot for fix it out; can I know that a usb webcam is Unable to open video device ???

ccrisan, 2018-01-26:
> @ccrisan There is a way to know ( with crontab or another way ) programatically that a usb webcam gives that error : Unable to open video device.

I'd be happy to know that way.