motioneye-project/motioneye #763: Disable thumb creation

ghubjules, 2018-01-13:
Is it possible to prevent the creation of thumbnail files?

I don't see a setting for it in the front end and adding `thumbnailer_interval 0` to motioneye.conf doesn't appear to disable it.

ccrisan, 2018-01-14:
It doesn't work that way but you could set it to a huge value and you should obtain the desired effect. But why do you want to disable thumbnails?

ghubjules, 2018-01-14:
I just don't need them for my application so would prefer to not have them created. I'll try setting a high value, thanks.

recrunchi, 2018-05-28:
Hi. 

excuse me, but i've tried the parameter thumbnailer_interval on motioneye.conf file, but syslog returns the following (after starting motioneye service):
"... motioneye[24281]: Starting motionEye server:WARNING:root:unknown configuration option: thumbnailer_interval"

It doesn't stop motioneye service from starting, and then i can see many thumb files created.

i'm on:
motionEye Version 0.39
Motion Version	 4.1.1
OS Version Raspbian 8.0 (jessie)

So, how can i disable thumb file creation? just dont wanna waste resources. i'm on raspberry pi3.
thanks.

ccrisan, 2018-05-28:
Actually the thumbnailer process no longer exists. Thumbnails are created as soon as the movie is generated by motion.

closed 2018-05-28 by ccrisan

zagrim, 2023-10-07:
I don't think there is any way to disable creation of thumbnails currently, it is hard-wired into movie_end event handling. You could of course create a clean-up job in the system to delete all .thumb files from the file storage path, but that's not very good solution, I know.

seizu, 2024-03-22:
> Thanks, I simply added this to the 'Run A Command' section in MontionEye and it works a treat:
> `rm -r /path/to/motioneye-data/*/*.thumb`

Good idea, thx.

rm -r stops if no .thumb is found, better to use

`rm -rf /path/to/motioneye-data/*/*.thumb`

or for any subfolder

`find /path/to/motioneye-data/ -type f -name "*.thumb"`