motioneye-project/motioneye #765: Unable to connect IP Camera

duckulaj, 2018-01-15:
I am trying to connect a Wansview W3 IP Camera but keep getting authentication failure.

[image]

[image]

[image]

[image]

I have tried also with authentication turned off for RTSP and still get authentication failure.

Also tried to add this as a simple MJPEG camera and stated unsupported camera type.

I changed the log level to debug, nothing in log file when trying to connect to the camera.

Any asistance would be appreciated

chrisf4lc0n, 2018-01-15:
The RTSP address is normally something like: rtsp://192.168.0.24:554/onvif1 or rtsp://192.168.0.24:554/onvif2 not just the IP address and port.

duckulaj, 2018-01-15:
Thank you very much for your fast response, all working now.

I used

rtsp://192.168.0.24:554/live/ch0

Thanks again

closed 2018-01-15 by duckulaj

kduytsch, 2018-07-13:
I know it's closed and I have another issue for authentication (using http). But I put a comment on this one as when I try to use the rtsp, I've got another error (and it looks like I have the same camera):

[image]

ewanyande, 2019-05-07:
I have installed motioneye on a PC with ubuntu but when I connect a camera it shows the "no signal sign". The same settings work well on motioneye on Raspberry. Please assist. I have attached a screenshot of the camera settings too.

[image]
[image]

chrisf4lc0n, 2019-05-07:
What Ubuntu Version?

ewanyande, 2019-05-07:
Hi,

I am using Ubuntu 16.04 LTS.

I have used the above script to allow traffic from camera and my laptop IP and still getting the same results. Also tried disabling the firewall and still not getting any results.

chrisf4lc0n, 2019-05-07:
Same Motion Version used on Pi and Ubuntu?

ewanyande, 2019-05-08:
Pi uses the following:

- motionEye Version | 0.39.3
- Motion Version | 4.1.1
- OS Version | motionEyeOS 20190119

The PC version user the following:
- motionEye Version | 0.35.1
- Motion Version | 3.2.12+git20140228

can one update the motion motion version to that of the Pi?

chrisf4lc0n, 2019-05-08:
`sudo pip motioneye --upgrade`
Install motion from the Wiki of MotionEye.