motioneye-project/motioneye #766: Continuous Recording ignores Working Schedule

homesmarty, 2018-01-15:
I am using continuous recording, but only need that during night. To not waste memory (and reduce wear the SD-card), I set working hours to 9:00 - 16:30 and chose "outside working schedule" - but continuous recording goes on and on. 
Ok, now I realized the setting is for during / outside working schedule is actually called "detect motion"... but it would make much more sense to also apply this to continuous recording.

ccrisan, 2018-01-16:
What you're asking here makes sense but is not how it was initially designed. The whole working schedule needs to be redesigned and reimplemented from scratch.

onshisan, 2018-03-30:
For me the current UI makes sense as-is inasmuch as it's the schedule for motion-triggered recording, but implementing an additional, separate switch and schedule for the "continuous" mode would handle this pretty well.

sigmarumberogen, 2018-07-16:
Hello, has there been any update on this? Or might be added in future updates? It will be really useful.

mcmidget, 2019-03-25:
+1

This functionality would be very useful for some setups as: recording *everything* during night or just during defined scheduled ranges.

maddust, 2019-05-23:
Any update on this feature? Would be great to have the option to schedule recordings without motion triggers. Maybe a separate option if it conflicts with  motion detection .

joseluissegura, 2019-07-31:
guys @homesmarty have you tried the option Settings -> WorkingSchedule (on) -> Detect Motion (Outside Working Schedule) ?

hoFFy84, 2019-09-19:
I believe I found a workaround:
In my situation I'm evaluating motion detection, which is not finally configured. In the meantime I thought about capturing still images every 2 seconds during the working schedule.
I set up the same camera two times: The first one is for motion detection, the second one is for the still images.
The one for the still images has motion detection enabled, but with these settings:
frame change threshold: 0,1
max change threshold 0
auto threshold tuning off
auto noise detection off
noise level 0
light switch detection 0
dewspeckle filter 0
motion gap 0
capture before 1
capture after 1
minimum motion frames 1
mask off

With these restrictive settings my Wansview W2 camera is producing enough noise (using a low resolution I get over 100000 changed pixles, which is much more than the 0,1% frame change threshold) to trigger the motion detection.
The creation of still images is set to motion triggered and is therefore only working during the schedule. 
The recorded movies are using the lowest possible quality and are auto-deleted after 1 day. This won't use too much disk space on my system.

I hope this will help some users who are looking for creating only still images during a working schedule.

zagrim, 2019-09-19:
If I understand your solution properly, that's actually pretty clever workaround, even though a bit tedious to set up.
For the record, you could also put `emulate_motion on` to "Extra Motion Options" to have motion detected continuously, without the need to set thresholds and noise levels artificially low.