motioneye-project/motioneye #768: Motioneye not start (status=1/failure)

Marthinius, 2018-01-17:
Hardware: Orange pi One
OS: Armbian server 16.04.3 LTS

After I've followed the guide the motioneye has been correctly installed, but I've a problem using webgui: http://[your_ip]:8765/   - nothing show....

command as root: **sudo systemctl status motioneye**

```
`● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset
   Active: failed (Result: exit-code) since Wed 2018-01-17 21:17:13 CET; 36min a
  Process: 491 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/mo
 Main PID: 491 (code=exited, status=1/FAILURE)

Jan 17 21:17:13 orangepione meyectl[491]:   File "/usr/local/lib/python2.7/dist-
Jan 17 21:17:13 orangepione meyectl[491]:     from tornado.web import Applicatio
Jan 17 21:17:13 orangepione meyectl[491]:   File "/usr/local/lib/python2.7/dist-
Jan 17 21:17:13 orangepione meyectl[491]:     from tornado import gen
Jan 17 21:17:13 orangepione meyectl[491]:   File "/usr/local/lib/python2.7/dist-
Jan 17 21:17:13 orangepione meyectl[491]:     from singledispatch import singled
Jan 17 21:17:13 orangepione meyectl[491]: ImportError: cannot import name single
Jan 17 21:17:13 orangepione systemd[1]: motioneye.service: Main process exited,
Jan 17 21:17:13 orangepione systemd[1]: motioneye.service: Unit entered failed s
Jan 17 21:17:13 orangepione systemd[1]: motioneye.service: Failed with result 'e
log file: tatus=1/FAILURE)...skipping...
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-01-17 21:17:13 CET; 36min ago
  Process: 491 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 491 (code=exited, status=1/FAILURE)

Jan 17 21:17:13 orangepione meyectl[491]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 29, in <mo
Jan 17 21:17:13 orangepione meyectl[491]:     from tornado.web import Application
Jan 17 21:17:13 orangepione meyectl[491]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 85, in <module>
Jan 17 21:17:13 orangepione meyectl[491]:     from tornado import gen
Jan 17 21:17:13 orangepione meyectl[491]:   File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 99, in <module>
Jan 17 21:17:13 orangepione meyectl[491]:     from singledispatch import singledispatch  # backport
Jan 17 21:17:13 orangepione meyectl[491]: ImportError: cannot import name singledispatch
Jan 17 21:17:13 orangepione systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 21:17:13 orangepione systemd[1]: motioneye.service: Unit entered failed state.
Jan 17 21:17:13 orangepione systemd[1]: motioneye.service: Failed with result 'exit-code'.
`
```
I have tried yet: **ps ax | grep motioneye**

`4970 pts/2    S+     0:00 grep --color=auto motioneye`

and **ps ax | grep motion**

`5370 pts/2    S+     0:00 grep --color=auto motion`

Finally this command: **sudo meyectl startserver -c /etc/motioneye/motioneye.conf** 

```
`Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 11, in <module>
    load_entry_point('motioneye==0.38', 'console_scripts', 'meyectl')()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
    import server
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 29, in <module>
    from tornado.web import Application
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 85, in <module>
    from tornado import gen
  File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 99, in <module>
    from singledispatch import singledispatch  # backport
ImportError: cannot import name singledispatch
`
```
I edit: **/etc/systemd/system/motioneye.service** and In Service section add User=root

Nothing helped, please help

chrisf4lc0n, 2018-02-06:
It looks like a problem with tornado, try to re-install it.

JamBalaya56562, 2026-11-07:
Closing, as this was the Python 2 release on Ubuntu 16.04, where the error pointed to Tornado, as said above. Current releases run on Python 3 and are installed with pip, which brings a matching Tornado, see the [README](https://github.com/motioneye-project/motioneye#installation). Please open a new issue with the output of `meyectl startserver` if the service still fails to start with a current release.

closed 2026-11-07 by JamBalaya56562