motioneye-project/motioneye #773: Feature Request - Auto create timelapse movies

phisher1, 2018-01-24:
An option to automatically create daily (or based upon interval) timelapse movies from motion detection images would be fantastic.

SealedJoy, 2018-09-13:
+1  - very practical feature to quickly skim through a summary of daily activity! could be pushed to users phone for viewing in the evening. 
I would love to see the ability to create timelapse images using this function across multiple days, this would open motioneye to the world of long term timelapse to see changes in nature.
This could be accomplished by adding the option to set a single folder that all images are saved too rather than a day by day splitting of photos into folders.

pkwentus, 2018-10-04:
I too would love this ability- and I'd like to be able to send that movie to a google drive etc.  
Reasoning:
Cloudy days tend to cause me havoc with tricking the system into making a lot of photos. (Yesterday, 3100) and today only 111 photos.  I tend to ignore these  large batches of photos because it takes FOREVER to review them.  I don't want to discount them though and a timelapse movie would be ideal.

Second note- if I try to run timelapse on this large photo set, the Rpi 3b+ will hang up.

tmjpugh, 2020-05-08:
Auto timelapse would be nice but even expose timelapse to API would be enough.  In this case it could be triggered externally vs external script to run with cron