motioneye-project/motioneye #774: Can't getting run 1-wire Temperatursensor

TheKitro, 2018-01-24:
Hi 
I'd like to read the temperatur with my motioneye Raspi too. I use the DS1820 1-wire sensor with 3 pins and 4,7kOhm resistor.
With an other Raspian raspi I tested it and it works. But under Motioneye it won't.
Under: /sys/bus/w1/devices/   i can't see any wire-1 devices.

I try the  
"w1-gpio"
"w1-therm"
in "/etc/modules" file.

and the 
"dtoverlay=w1-gpio,gpiopin=4" 
addition in the "/boot/config.txt" file.

But only on the Motioneye I don't get it running. I used beside other tutorials this "http://www.netzmafia.de/skripten/hardware/RasPi/Projekt-Onewire/index.html" tutorial. 

Is there something special to know on Motioneye?

Regards Kitro

chrisf4lc0n, 2018-01-25:
Are you talking abou MotionEyeOS?

TheKitro, 2018-01-26:
yes

chrisf4lc0n, 2018-01-27:
I think you have answered your own question.
MotionEyeOS is not Raspbian.
https://github.com/ccrisan/motioneyeos/wiki

chrisf4lc0n, 2018-01-27:
If you want the full functionality, you will need to install Raspbian and the MotionEye on top of it.

TheKitro, 2018-01-27:
ok I thought that can be installed after

chrisf4lc0n, 2018-02-04:
@ccrisan I think we can close this one down.

closed 2018-02-05 by ccrisan