motioneye-project/motioneye #776: Installation error on CentOS7

jwc866, 2018-01-26:
Hello:
I am attempting to install motioneye on a CentOS 7 box which has Python 2.7 by default.  I have the dependencies installed: python-pip python-devel gcc libcurl-devel libcurl libjpeg.  However when I run "pip install motioneye", it downloads the extra packages and eventually fails after running setup.py, with the error:

 TypeError: 'NoneType' object is not iterable

    ----------------------------------------

Command "/usr/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-N20aUI/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-kfYb9p-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-N20aUI/pillow/

Any ideas on what I can check?  Thank you for the help.

ccrisan, 2018-01-26:
Please attach the full setup log. There's no way I can tell which package generated this error from your single log line.

jwc866, 2018-01-26:
I just uploaded the full log.  Thanks very much for your help.

[motioneye_install_log.txt](https://github.com/ccrisan/motioneye/files/1668847/motioneye_install_log.txt)

ccrisan, 2018-01-28:
Your problem actually seems to be that python package `pillow` fails to install. A quick googling led me to https://github.com/python-pillow/Pillow/pull/2951. It may or may not be related and you may or may not get away with specifying a different version of pillow.

In any case, this is not a valid motionEye issue. Please make sure that you have installed `pillow` python package on your system. If you simply can't do that, please open a `pillow` issue.

closed 2018-01-28 by ccrisan