motioneye-project/motioneye #778: Integrate with free machine learning based image analysis to reduce false positives

MagnusThome, 2018-01-28:
A problem with motion detection is that you do get false positive notifications. But there *is* a solution.

A guy named Mark West has put together a superb system using Amazon AWS micro services to send all motion detection images through a machine learning cloud based image analysis. This way he can remove false positives and only mail notifications depending on what kind of objects that actually are visible.

He's written a nice howto here https://www.bouvet.no/bouvet-deler/utbrudd/smarten-up-your-pi-zero-web-camera-with-image-analysis-and-amazon-web-services-part-1

I set it up for myself this afternoon without any previous knowledge of AWS and it works great so far.

-----------------------

Since Marks system is 100% Amazon based the storage solution is of course Amazon S3.

So there is one improvement that could be done to MotionEye to integrate with Marks setup. And that is native support in MotionEye to upload images to Amazon S3, in addition to the FTP, sftp, google drive and dropbox options currently already available. 

There are a whole number of other ways to get the files uploaded to S3 so it's not a show stopper problem. But IF it isn't a huge job to add it (the MotionEye upload code looks very nicely modularized) it would be very nice and maybe it would get more people to use image analysis, resulting in a tremendously improved system as a whole!

I'll take some time to go through the file uploading code but I'm not sure I can manage this by myself. We'll see :-D

MagnusThome, 2018-01-29:
For the time being I use this script to upload:

```
#!/usr/bin/env bash

#S3 parameters
S3KEY="xxxxxxxxxxxxxxxxxx"
S3SECRET="xxxxxxxxxxxxxxxxxxxx"
S3BUCKET="xxxxxxxxxxxxxxxxxxx"
S3STORAGETYPE="STANDARD" #REDUCED_REDUNDANCY or STANDARD etc.
AWSREGION="s3-xxxxxxxxxx"

function putS3
{
  path=$1
  file=$2
  aws_path="upload"
  bucket="${S3BUCKET}"
  date=$(date +"%a, %d %b %Y %T %z")
  acl="x-amz-acl:private"
  content_type="application/octet-stream"
  storage_type="x-amz-storage-class:${S3STORAGETYPE}"
  string="PUT\n\n$content_type\n$date\n$acl\n$storage_type\n/$bucket/$aws_path/$file"
  signature=$(echo -en "${string}" | openssl sha1 -hmac "${S3SECRET}" -binary | base64)
  curl -s -X PUT -T "$path/$file" \
    -H "Host: $bucket.${AWSREGION}.amazonaws.com" \
    -H "Date: $date" \
    -H "Content-Type: $content_type" \
    -H "$storage_type" \
    -H "$acl" \
    -H "Authorization: AWS ${S3KEY}:$signature" \
    "https://$bucket.${AWSREGION}.amazonaws.com/$aws_path/$file"
}

putS3 "/var/lib/motioneye/Camera1/" "lastsnap.jpg"
```

JamBalaya56562, 2026-11-09:
Upload to Amazon S3 was added in #1063 and is there since 0.43.1, so images can go straight to a bucket for that kind of analysis. Image analysis itself is outside motionEye. Closing.

closed 2026-11-09 by JamBalaya56562