motioneye-project/motioneye #779: Example Config is IPv4-only, yet tornado supports IPv6

cvmiller, 2018-01-29:
Given this day in age where IPv4 addresses are exhausted, and 20% of the traffic on the internet is IPv6, it would be great to see MotionEye support IPv6 as well. 

Looking into the code a bit, it appears that a small change to the settings.py will enable MotionEye to support IPv6 (at least for the control webserver). The current settings.py has:

```
# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
LISTEN = '0.0.0.0'

```

Yet, setting LISTEN to an empty string will enable the tornado webserver to listen on both IPv4 AND IPv6

```
# the IP address to listen on
# (0.0.0.0 or :: for all interfaces, 127.0.0.1 or ::1 for localhost, blank for both IPv4 & IPv6)
LISTEN = ''
```

At a minimum it would be nice if the comments reflected support for IPv6, and perhaps the default settings.py could be a blank string, enabling both IPv4 AND IPv6 support for your fine software.

ccrisan, 2018-01-30:
Makes sense, as long as this is strictly motionEye-related and has nothing to do with motionEyeOS, where IPv6 is currently disabled.

cvmiller, 2018-01-30:
Great. IPv6 is the future of the internet (all growth will be using IPv6). MotionEye is a great piece of software, and it will be even better with IPv6 support.

ttimasdf, 2018-12-22:
besides the hack @cvmiller suggests, 
[meyectl](https://github.com/ccrisan/motioneye/blob/d77d3d81286e9fe4d25800d19df730f59c52e6e1/motioneye/meyectl.py#L74) won't accept a line with empty value, because the config parser is a simple strip and split. It does not recognize quotes as well. Use standard library `configparser` is easier and better IMHO, but at the time being we could just add another hack to the whole giant hacks `meyectl.py`:

```python
# ...
             elif value.lower() == 'true':
                 value = True

             elif value.lower() == 'false':
                 value = False

             # here's the hack
             elif value in ['""', "''"]:
                 value = ""

             elif isinstance(curr_value, int):
                 value = int(value)

             elif isinstance(curr_value, float):
# ...
```
Then, tornado listens dual stack as expected 😏

cvmiller, 2020-08-23:
Thanks @ttimasdf, I went with the simpler "fix" of making the change in `settings.py`, and it works well. I now can manage motioneye web interface over IPv6, as well as see the video stream.

This "fix" also applies to motioneyeos (which now has IPv6 enabled, but not in the config). Including cross link to that issue:

https://github.com/ccrisan/motioneyeos/issues/802

DrCR77, 2020-12-11:
Hi @cvmiller @ttimasdf !
Thanks for this but I am a bit confused as to what fix to apply now. I cannot access neither settings.py nor meyectl.py directly. My ISP is on ipv6 now, so I have to make it work under ipv6 somehow and appreciate your help on this. Thanks!

cvmiller, 2020-12-12:
Excellent timing! I have been meaning to write this up as a short article. Because my motioneyeOS SBC is inside my house, I chose to take the easy way, and only edit the settings.py file. 

The write-up can be found at:

http://www.makiki.ca/ipv6/motioneyeos_and_ipv6.html

starbasessd, 2020-12-12:
Bad link?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cvmiller, 2020-12-12:
Sorry, I assumed you were using IPv6. www.makiki.ca is an IPv6-only webserver.

Here's a dual stack link:
http://www.makikiweb.com/ipv6/motioneyeos_and_ipv6.html

starbasessd, 2020-12-12:
Very nice! Could be helpful for other situations, too.
Unfortunately (or fortunately), I live in an area where the ISP is still
NAT'ing to the IPv4 space.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cvmiller, 2020-12-13:
Unfortunately, since most of the world has run out of IPv4 address space, ISPs running CGNAT (Carrier Grade NAT) is going to be more and more common. A good reason to move to IPv6, where there is no NAT.

starbasessd, 2020-12-13:
True, but I hope they bring back an IPv6 version of NAT. To have truly
unique addresses that can be tied to you is not good for security. There
would no longer be any anonymity. But that is off-topic and a discussion
for another day...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DrCR77, 2020-12-14:
> http://www.makiki.ca/ipv6/motioneyeos_and_ipv6.html

Many thanks! 
My ISP switched to ipv6 native + DS Lite that means a lot of trouble. It was all much easier in the ipv4 world, not so speak of access from mobile phone which will not be possible at all (ipv4 only).

Anyway, I tried to follow the guide and used Filezilla to transfer the files (scp didn't work). 
As a brief remark, found a little typo here: 
uncompyle6 -o settings.pyc
better: uncompyle6 -o . settings.pyc

That settings.py adjustment worked in the end. 

Unfortunately, I still cannot access the camera. I guess it is because I cannot find the right IPv6 GUA address. In your example there is a global dynamic, similar to my result. So I guess you used the second ipv6 address (your example fe80::ba27:ebff:fe6c:388) to be used as GUA? Despite an open port (exposed host) on my router I cannot ping6 it. 

As to the AAAA record: I added this address to my domain but as expected without any response. 
If it all works with v6 end to end there is no further need for dyndns or am I wrong?

cvmiller, 2020-12-14:
Thanks for catching the typo. I have corrected the page.

It doesn't sound like you have the "right" IPv6 address for MotioneyeOS.  As you noticed, it is normal to have multiple IPv6 addresses on an interface. A GUA will start with '2' (that is the only number allocated at this time). The 'fe80' address is a link-local address, and will not cross a router.

If your laptop is  on the same LAN as MotioneyeOS SBC, then you should be able to ping (or ping6) the GUA. You should also be able to ssh to the SBC using the GUA (if you enabled that). 

If you want to contact me directly (click on my name to find my email), we can discuss the specifics of your network.

SimonLouis80, 2021-03-21:
@cwmiller:  I've tried your solution on http://www.makikiweb.com/ipv6/motioneyeos_and_ipv6.html, but it's not working for me.
sshd is working with ipv6, but motioneye still not reachable on ipv6. 
I have double checked configuration in settings.py (which i have uncomplied also within an alpine linux container - like you described). Not sure why motioneye not responsible on ipv6-address.

ttimasdf said:  "If you choose not to patch meyectl.py, you MUST comment out the listen line in config file for it will override the one specified in settings.py"
--> Do i need to remove listen line from motioneye.conf ? 

I tried, but still working for me.

Thanks for your reply!

starbasessd, 2021-03-21:
An issue here:
-----
Fortunately, the distro does include openssh, and it is possible to scp files off and back onto the SBC. First one needs to enable IPv6 for ssh. Edit /etc/ssh/sshd_config and add the line:

ListenAddress ::

and restart sshd

/etc/init.d/S60sshd restart

Now you should be able to ssh in on the IPv6 address of MotioneyeOS. How did you get the address. The easiest way is to make a note of it while you were in editing the sshd config file with the ip addr command.
-----
MotionEyeOS is mostly set to Read Only. The /etc/ tree is. I do not understand why you didn't get an error trying to upload the file back to the motionEyeOS.
You would have to either switch Enable Debugging ON or mount the SDCard in a device that understands ext4 and doesn't honor permissions or ro status.
Ditto with the /usr/ tree. With the Overlay file system, both exist in /data/ and could be modded there, too. /data/usr/ and /data/etc/

cvmiller, 2021-03-21:
Sorry, My documentation was incomplete. I have updated it. One needs to remount the root system as RW.

```
mount -o remount,rw /
```

cvmiller, 2021-03-21:
> ttimasdf said: "If you choose not to patch meyectl.py, you MUST comment out the listen line in config file for it will override the one specified in settings.py"
> --> Do i need to remove listen line from motioneye.conf ?

Checking my notes, I don't see that I did this (removing listen line from config file), but I probably did. Did you restart mtioneye after all your changes?

Unfortunately, I am no longer running motioneye, so I can't easily check the config file.

SimonLouis80, 2021-03-21:
Yes i did a restart. 

This is my uncomplied settings.py, which i have copied from alpine linux back to motioneye.
I copied it to the folder /usr/lib/python2.7/site-packages/motioneye/, renamed the settings.pyc to settings.pyc.bak and restarted motioneye. After the reboot the new settings.pyc has been created.

***** settings.py ******************
import logging, os.path, sys, socket, motioneye
config_file = None
PROJECT_PATH = os.path.dirname(motioneye.__file__)
TEMPLATE_PATH = os.path.join(PROJECT_PATH, 'templates')
STATIC_PATH = os.path.join(PROJECT_PATH, 'static')
CONF_PATH = [
 sys.prefix, ''][(sys.prefix == '/usr')] + '/etc/motioneye'
for d in ['/run', '/var/run', '/tmp', '/var/tmp']:
    if os.path.exists(d):
        RUN_PATH = d
        break
else:
    RUN_PATH = PROJECT_PATH

for d in ['/log', '/var/log', '/tmp', '/var/tmp']:
    if os.path.exists(d):
        LOG_PATH = d
        break
else:
    LOG_PATH = RUN_PATH

MEDIA_PATH = '/var/lib/motioneye'
LOG_LEVEL = logging.INFO
LISTEN = ''
PORT = 8765
MOTION_BINARY = None
MOTION_CONTROL_LOCALHOST = True
MOTION_CONTROL_PORT = 7999
MOTION_CHECK_INTERVAL = 10
MOTION_RESTART_ON_ERRORS = False
MOUNT_CHECK_INTERVAL = 300
CLEANUP_INTERVAL = 43200
REMOTE_REQUEST_TIMEOUT = 10
MJPG_CLIENT_TIMEOUT = 10
MJPG_CLIENT_IDLE_TIMEOUT = 10
SMB_SHARES = False
SMB_MOUNT_ROOT = '/media'
WPA_SUPPLICANT_CONF = None
LOCAL_TIME_FILE = None
ENABLE_REBOOT = False
ENABLE_UPDATE = False
SMTP_TIMEOUT = 60
LIST_MEDIA_TIMEOUT = 120
LIST_MEDIA_TIMEOUT_EMAIL = 10
ZIP_TIMEOUT = 500
TIMELAPSE_TIMEOUT = 500
ADD_REMOVE_CAMERAS = True
VALIDATE_CERTS = True
PASSWORD_HOOK = None
HTTP_BASIC_AUTH = False
SERVER_NAME = socket.gethostname()
******

SimonLouis80, 2021-03-21:
netstat shows:

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:139             0.0.0.0:*               LISTEN      1084/smbd
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1144/python
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN      1160/motion
tcp        0      0 0.0.0.0:21              0.0.0.0:*               LISTEN      1068/proftpd: (acce
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1046/sshd
tcp        0      0 0.0.0.0:445             0.0.0.0:*               LISTEN      1084/smbd
tcp        0      0 127.0.0.1:7999          0.0.0.0:*               LISTEN      1160/motion
tcp6       0      0 :::139                  :::*                    LISTEN      1084/smbd
tcp6       0      0 :::22                   :::*                    LISTEN      1046/sshd
tcp6       0      0 :::445                  :::*                    LISTEN      1084/smbd
[root@birds motioneye]#

cvmiller, 2021-03-21:
> you MUST comment out the listen line in config file for it will override the one specified in settings.py"

Did you remove the "listen" line in the /etc config file? As stated, it will *over-ride* your settings.py file.

SimonLouis80, 2021-03-21:
@cvmiller. Now it`s working! Was my mistake!
There are two motioneye.conf files. One at /etc and the other at /data/etc/
I removed the lines from the files in /etc. But it must be removed from the file located in /data/etc/

:-)

cvmiller, 2021-03-21:
Hey, that is great news. I have updated my web page with that info, and hopefully future folks will not run into this issue.

cvmiller, 2025-07-22:
No, you still need to make the manual changes (above).

closed 2026-11-16 by MichaIng